OE4T/meta-tegra #1698: Issues flashing Jetson Orin nano devkit with TEGRA_SIGNING_ARGS set and fused keys

wavesid, 2024-09-18:
Hello

**Describe the bug**
I have the exact same setup as this issue: https://github.com/OE4T/meta-tegra/issues/1639
i already opened an issue and the outcome was to fuse the keys and re-test https://github.com/OE4T/meta-tegra/issues/1674.

I am using Jetson Orin Nano 8GB devkit, with this option: `TEGRA_SIGNING_ARGS` set to the following

```
TEGRA_SIGNING_ARGS = "-u pkc.key -v sbk.key"
```

**The keys are fused.
The official script `l4t_initrd_flash.sh` from Jetson Linux works with my keys so I suppose the fusing happened without issues and the keys are correct.**

This is the logs:

using `sudo ./doflash.sh`

```
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0100 ] Parsing partition layout
[   0.0105 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0119 ] Parsing partition layout
[   0.0122 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0134 ] mb1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --mb1_bin
[   0.0134 ] psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --psc_bl1_bin
[   0.0134 ] Boot Rom communication
[   0.0137 ] tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
[   0.0141 ] BR_CID: 0x80012344705DF11F2400000013028100
[   0.0405 ] Sending bct_br
[   0.0800 ] Sending mb1
[   0.0807 ] ERROR: might be timeout in USB write.
Error: Return value 3
Command tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
```

using `sudo ./initrd-flash`

```
Starting at 2024-08-23T18:03:47+02:00
Machine:       jetson-orin-nano-devkit-nvme
Rootfs device: nvme0n1p1
Found Jetson device in recovery mode at USB 1-1
== Step 1: Signing binaries at 2024-08-23T18:03:47+02:00 ==
== Step 2: Boot Jetson via RCM at 2024-08-23T18:03:48+02:00 ==
Found Jetson device in recovery mode at USB 1-1
./initrd-flash: line 191: ./rcm-boot.sh: No such file or directory
ERR: RCM boot failed at 2024-08-23T18:03:48+02:00
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'scarthgap' (latest commit) with `MACHINE` set to 'jetson-orin-nano-devkit-nvme'
2. Set `TEGRA_SIGNING_ARGS` with `-u pkc.key -v sbk.key`
3. Build with bitbake image
4. Deploy to hardware with method tegraflash using `sudo ./doflash.sh` (or using `sudo ./initrd-flash`)
5. See logs above

**Additional context**
I checked USB connection, using `PKC + SBK` keys, I do not have logs in UART

**The keys are fused.
The official script `l4t_initrd_flash.sh` from Jetson Linux works with my keys so I suppose the fusing happened without issues and the keys are correct.**

Let me know if there is any way to debug

ichergui, 2024-09-19:
Hi @wavesid 

Did you try to build without `TEGRA_SIGNING_ARGS` and flash your device with this command line `$ sudo ./doflash.sh -u PKC.pem -v SBK.key` ?

wavesid, 2024-09-19:
Yes if I build an image without `TEGRA_SIGNING_ARGS` set, both commands works without issues:

`sudo ./initrd-flash -u PKC.pem -v SBK.key` and `sudo ./doflash.sh -u PKC.pem -v SBK.key`

ichergui, 2024-09-19:
@wavesid  thanks for your quick feedback. I need to investigate that.I tried the `Orin AGX devkit` and it does work perfectly. I mean signing at build time and post build. Both methods works fine.
@madisongh Could you please share your thoughts ? 
@wavesid is able to flash his device when signing/encrypting images/binaries post build. with the following command 

```
$ sudo ./doflash.sh -u PKC.pem -v SBK.key
```
However, the signing at build time doesn't work when using `jetson-orin-nano-devkit-nvme`.

madisongh, 2024-09-19:
If it works when directly flashing, but not when signing during the build, then my bet would be on some difference between the default FAB/BOARDSKU/etc. settings we have for the machine vs. the actual ones being reported during the flashing process.

ichergui, 2024-09-24:
@wavesid Any update about this issue ? 
Were you able to check the variables mentioned by @madisongh ?

madisongh, 2024-09-26:
Could be the issue I mentioned in #1668

tgagneret-embedded, 2024-09-30:
Hello,

```
[   0.0800 ] Sending mb1
[   0.0807 ] ERROR: might be timeout in USB write.
Error: Return value 3
```

I had the same issue because of USB autosuspend. Run the following command to disable usb autosuspend.
`echo -1 >  /sys/module/usbcore/parameters/autosuspend`

I'm not sure it will solve your issue, but I hope it helps :)

wavesid, 2024-10-01:
I will perform some more tests this week and keep you updated

Thanks

akfaro, 2025-01-10:
I also observe this symptom with an Orin NX board, when trying to flash signed binaries with .`initrd-flash`

I work with meta-tegra at scarthgap, commit 5c5de2f1b46ebe584ad46195b2083ed1157b905d from Nov 1 2024, so without the fix https://github.com/OE4T/meta-tegra/issues/1668. 
However, I don't think that this issue is related. The mentioned fix addresses wrong binaries being sent to device, this issue is about device not even starting in RCM when working with `initrd-flash`.

I see why the script fails where it does, but not how to fix this:

As expected, providing a key via `TEGRA_SIGNING_ARGS` created a `.presigning-vars` file in the `tegraflash` package, thus setting `PRESIGNED=yes` within `initrd-flash`, which in turn leads to
* "Step 1: Signing binaries" being skipped,
* subfolder `rcmboot_blob/` with a `rcmbootcmd.txt` file in it not being created, and 
* the file `./rcm-boot.sh` not being generated while unconditionally required by the function `wait_for_rcm`.

Or did I miss something, and non-existent `./rcm-boot.sh`  should not be an issue with a more recent `meta-tegra`?

madisongh, 2025-01-11:
The problem is that NVIDIA changed the way its scripts handle signing the RCM boot blob in L4T R36.4, and during the upgrade to that version. I had updated our scripts to account for that change for normal, non-secured devices, but I don't have a secured Orin myself, so I missed out on the needed changes to handle that case.

Sorry about that. I was counting on others who have secured devices to help find issues like these, but this use case got missed.

I've got PR #1800 pending that should take care of this.  I can't fully test it, unfortunately, but it doesn't break the non-secured use case, and it looks like it generates usable binaries when I do signing during the build.

closed 2025-01-12 by madisongh

madisongh, 2025-01-12:
I've ported the changes from #1800 to the other R36-based branches.  Let me know if they work for you.

madisongh, 2025-01-19:
See also #1815

otiv-arne-janssens, 2025-01-27:
I have exactly the same issue on an orin nx, but unfortunately, I don't have the luxury to update to R36.4.0/JetPack 6.1
Attempting to cherry-pick the solution in #1800 has failed so far. Any ideas on where to go from here?

PR #1800 (2025-01-12): tegra-helper-scripts: fix RCM boot blob signing
With L4T R36.4, the odmsign.func script now supports signing the RCM boot blob used for the initrd-based flashing process. While the initrd-flash script was updated to handle this, more changes are needed to handle signing all binaries during the build.

* Update the flash helper script to run signing on the RCM boot blob when needed.

* Update the initrd-flash script to drop the extra helper invocation for signing the RCM boot blob, since that is now covered with the normal signing step for the internal flash. And when pre-signed, initrd-flash j [... cut]