OE4T/meta-tegra #1704: Compiler mismatch building `nvidia-kernel-oot` in workspace generated from devtool

arrow53, 2024-09-23:
**Describe the bug**
When I build nvidia-kernel-oot in a workspace I get a compiler mismatch. I'm using `tegra-demo-distro`

| The major and minor number of the compiler used to
| compile the kernel:
| 
| x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38
| 

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'scarthap' with `MACHINE` based on 'jetson-orin-nano-devkit'
2. Build with bitbake argument 'devtool modify nvidia-kernel-oot' then `bitbake  nvidia-kernel-oot`

```
| make[1]: warning: -j0 forced in submake: resetting jobserver mode.
| 
| Warning: Compiler version check failed:
| 
| The major and minor number of the compiler used to
| compile the kernel:
| 
| x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38
| 
| does not match the compiler used here:
| 
| aarch64-oe4t-linux-gcc (GCC) 13.3.0
| Copyright (C) 2023 Free Software Foundation, Inc.
| This is free software; see the source for copying conditions.  There is NO
| warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
| 
| 
| It is recommended to set the CC environment variable
| to the compiler that was used to compile the kernel.
| 
| To skip the test and silence this warning message, set
| the IGNORE_CC_MISMATCH environment variable to "1".
| However, mixing compiler versions between the kernel
| and kernel modules can result in subtle bugs that are
| difficult to diagnose.
| 
| *** Failed CC version check. ***
```

ichergui, 2024-09-24:
Hi @arrow53 
Could you please elaborate about the change that you are trying to apply in `nvidia-kernel-oot` ?

ichergui, 2024-09-24:
Thanks @madisongh for the PR.
@arrow53 Could you please check with this fix and keep us posted ? 
I will close this issue based on your feedback

madisongh, 2024-09-24:
You can also work around the issue without pulling in this fix by doing a `git restore -- Makefile` in the devtool workspace, then re-running the build.

arrow53, 2024-09-24:
@madisongh your fix did the trick. thanks for the quick turn on this.

PR #1705 (2024-09-24): nvidia-kernel-oot: conditionalize Makefile copying for do_unpack
When using devtool, the first bitibake build after the devtool workspace is set up triggers re-execution of our do_unpack postfunc, which is overwriting the patched copy of the top-level makefile.

Avoid this by copying it in only if the top-level makefile isn't already there.

Fixes #1704
closed 2024-09-24 by arrow53