OE4T/meta-tegra #1706: Support for Orin NX 16 GB affected by PCN 211361 (kirkstone)

graceagrace, 2024-09-24:
Hi there,

Is it on the roadmap to patch in L4T 35.6 to the `kirkstone` branch of meta-tegra? I recently became aware of PCN 211361 (a Micron DRAM change on the Orin NX 16 GB), which requires updates to the BCT and DVFS. So far, I've been unsuccessful in getting my distributor to provide a patch with only the necessary changes to support the new DRAM.

madisongh, 2024-09-24:
Yes, that's the plan. We'll be porting the 35.6 update back to `kirkstone` after it goes into the `scarthgap-l4t-r35.x` branch. Don't have a specific timeframe for that, but hopefully within the next several weeks.

graceagrace, 2024-09-24:
Excellent, thank you!

madisongh, 2024-10-11:
Both the `scarthgap-l4t-r35.x` and `kirkstone` branches have been updated to L4T R35.6.

graceagrace, 2024-10-11:
Great, thank you for the update and the hard work on this!

closed 2024-10-11 by graceagrace