OE4T/meta-tegra #1707: Add new mainline kernel driver to TX2

Adnan-Elhammoudi, 2024-09-24:
Hello,

Is there any way to add a new mainline kernel driver (max96714) to the layer? I am not able to do so because the kernel version used for TX2 is 4.9. I tried to avoid backporting the driver and instead used the backport project, but it did not solve the issue as the latest version was 6.1

I would appreciate any suggestions.

dwalkes, 2024-09-24:
@Adnan-Elhammoudi AFAIK I don't believe this is possible for Jetpack 4/TX2 targets.  I think you'll need to backport to the nvidia supported kernel release.

Adnan-Elhammoudi, 2024-09-25:
@dwalkes Thank you for your input. 

How do you think the best approach would be to tackle this? Do you have any recommendations or specific steps to follow for the backporting process on the TX2 platform?

madisongh, 2024-09-26:
There's nothing necessarily specific to the TX2, but the kernel version, as Dan pointed out, is 4.9, not 5.15.

You could check with your vendor/reseller to see if they've already done (or know of) a backport for that version. Otherwise, it's a matter of understanding the hardware well enough, and kernel driver development well enough, to implement and debug the driver yourself.