OE4T/meta-tegra #1724: [Jetson Orin Nano] Device-tree overlays are not applied

Galadrin, 2024-10-15:
# Bug Description

We are attempting to deploy a Yocto distribution on an Nvidia Jetson Orin Nano devkit. As part of this, we need to utilize the 40-pin header. To ensure the system boots with the correct pinmux configuration, we aim to apply the `tegra234-p3737-0000+p3701-0000-hdr40.dtbo` as a device tree overlay.

We followed the instructions in the [meta-tegra documentation](https://github.com/OE4T/meta-tegra/wiki/Using-device-tree-overlays#runtime-application-of-overlays) to apply this overlay. 

In the `conf/local.conf` file, we added the following configuration:
``` 
UBOOT_EXTLINUX_FDT = "/boot/${DTBFILE}"
TEGRA_PLUGIN_MANAGER_OVERLAYS:append = " tegra234-p3737-0000+p3701-0000-hdr40.dtbo "
```

When running the command:
```
bitbake core-image-base -e
```
We confirmed that the overlays are set up as follows:
```
# $TEGRA_PLUGIN_MANAGER_OVERLAYS [3 operations]
#   :append /home/dapie/orin-yocto/build/conf/local.conf:257
#     " tegra234-p3737-0000+p3701-0000-hdr40.dtbo "
#   set /home/dapie/orin-yocto/sources/openembedded-core/../meta-tegra/conf/machine/include/tegra-common.inc:19
#     [_defaultval] ""
#   set? /home/dapie/orin-yocto/sources/openembedded-core/../meta-tegra/conf/machine/include/orin-nano.inc:30
#     "tegra234-carveouts.dtbo tegra-optee.dtbo tegra234-p3768-0000+p3767-0000-dynamic.dtbo"
# pre-expansion value:
#   "tegra234-carveouts.dtbo tegra-optee.dtbo tegra234-p3768-0000+p3767-0000-dynamic.dtbo tegra234-p3737-0000+p3701-0000-hdr40.dtbo "
TEGRA_PLUGIN_MANAGER_OVERLAYS="tegra234-carveouts.dtbo tegra-optee.dtbo tegra234-p3768-0000+p3767-0000-dynamic.dtbo tegra234-p3737-0000+p3701-0000-hdr40.dtbo "
#
```

# Issue

After booting, we inspected the device tree but could not find the overlay entries. Further investigation revealed that even predefined overlays, such as tegra234-carveouts.dtbo and tegra-optee.dtbo, are not being applied.

# Steps to Reproduce

1. Build the meta-tegra Yocto image with MACHINE set to jetson-orin-nano-devkit.
2. Use `bitbake core-image-base` to build the image.
3. Deploy the image to hardware:
```
mkdir -p /tmp/tegraflash
tar -C /tmp/tegraflash -xavf /path/to/core-image-base-jetson-orin-nano-devkit.rootfs.tegraflash.tar.gz
cd /tmp/tegraflash
sudo ./dosdcard.sh /dev/mmcblk0 -y
```
4. boot and login to the board
5. Check for the pinmux overlay entries using:
```
ls /proc/device-tree/bus\@0/pinmux\@2430000/exp-header-pinmux
```

madisongh, 2024-10-19:
I may be misremembering, but I'm pretty sure that when you use TEGRA_PLUGIN_MANAGER_OVERLAYS, the overlays are attached to the end of the device tree that's embedded in the UEFI bootloader image. That's all done through the magic of NVIDIA's tegraflash.py script. You would have to update your SPI flash, and not just your SDcard, so the bootloader gets updated, to use this technique.

tschoop, 2024-10-21:
I can confirm this. DT overlays are stored in one of the partitions in the QSPI NOR flash. So to bring changed  DT overlays onto a device you have to do a complete low level flash (initrd-flash) or you have to do a UEFI capsule update.

Galadrin, 2024-10-21:
ok thanks, 
indeed, the Overlay is appended at the end of the DTB,
thanks to the `UBOOT_EXTLINUX_FDT` variable, the DTB is installed on my SDCard and used by the bootloader.

I finally created a new device tree without the use of overlays.

closed 2024-10-23 by madisongh