OE4T/meta-tegra #1773: Booting Jetson AGX Orin devkit with Jetpack 6.1

sapiippo, 2024-11-25:
**Describe the bug**
After updating to Jetpack 6.1 (on master branch), I'm unable to get the Jetson AGX Orin devkit to boot with the tegraflash image.

Flashing goes trough without errors, but after device goes into boot loop with last output as 

```
Jetson System firmware version v36.4.0 date 2024-10-02T18:07:54+00:00
ESC   to enter Setup.
F11   to enter Boot Manager Menu.
Enter to continue boot.
......����
Shutdown state requested 1
Rebooting system ...
```

**Additional context**

The default Ubuntu image from NVIDIA's SDK Manager boots correctly.

dwalkes, 2024-11-25:
Have you attempted to reproduce on [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro) with demo-image-base or similar?  If not I'd start there and see if you can reproduce.  If you can't you can compare this setup to your layer to help determine any relevant differences.

sapiippo, 2024-11-27:
I got same result with meta-tegra-distro and demo-image-base.
flashed with initrd-flash, device ends up in boot loop.

ichergui, 2024-11-27:
Could you please explain why you are using `initrd-flash` ? 
`initrd-flash` is recommended to be used when having external device e.g. NVMe 
Jetson AGX Orin have eMMC. So, please use `doflash.sh` script

ichergui, 2024-11-27:
Unrelated subject. The comment you shared is about Orin NX.

sapiippo, 2024-11-28:
```
Parsing module EEPROM:
--- Parsing board ID (3701) succeeded.
--- Parsing board version (500) succeeded.
--- Parsing board SKU (0000) succeeded.
--- Parsing board REV (J.0) succeeded.
jetson-agx-orin-devkit found.
```
This is from the flashing logs of NVIDIA's L4T image.

`jetson-agx-orin-devkit` machine config is still working for this board from `scarthgap-l4t-r35.x` branch.
The board is a stock item from NVIDIA. I'm not sure about NVMe, does that make a difference?

Unfortunatly, I won't have access to the device for a while, so cannot get logs from UEFI. Will do that at next opportunity.

sapiippo, 2024-12-09:
So, it turned out someone had installed NVMe disk to the Orin board and that was causing the issue. Same thing happens also with Jetpack 5 based images.

Changing the boot order doesn't help when NVMe disk is attached.

Attached is log from demo-image with UEFI logging enabled.
[demo_with_nvme_not_booting.log](https://github.com/user-attachments/files/18063326/demo_with_nvme_not_booting.log)

madisongh, 2024-12-09:
How are you changing the boot order?  Through the UEFI menus? That *should* work, assuming the ESP on the eMMC is valid, and that you save the boot order change before exiting UEFI.

Oh, but the script in tegra-minimal-initrd, by default, just looks for a partition named `APP` to boot from, so if you have an APP partition on the NVMe, it could mount that one instead of the eMMC one. The logic in there could probably be improved. So UEFI could be loading the kernel off the eMMC, but the initrd would still picking up the wrong rootfs.

You could customize the script to avoid this, or set `KERNEL_ARGS:append = " root=/dev/mmcblk0p1"` to override the default in the initrd scripts, or remove the `APP` partition from your NVMe if you don't need it.  There's also an `--erase-nvme` option on `initrd-flash` that can be used to wipe the entire NVMe drive during flashing.

ichergui, 2025-01-26:
Hi @sapiippo 
Any update on this issue ?

sapiippo, 2025-01-28:
> Oh, but the script in tegra-minimal-initrd, by default, just looks for a partition named APP to boot from, so if you have an APP partition on the NVMe, it could mount that one instead of the eMMC [...]

This seemed to be the cause. Erasing the NVMe or removing it solved the booting issue.
I did not try to customize the initrd script.

closed 2025-07-09 by madisongh