OE4T/meta-tegra #1784: cuda-samples is not buildable with latest master with cuda 12.6

kraj, 2024-12-09:
**Describe the bug**
cuda has been upgraded on master to 12.6 but the samples repo is still pinned at 12.5 release
This maybe the reason for non-buildable cuda-samples recipe

```
| NOTE: make -j 44 GENCODE_FLAGS=--gpu-architecture=compute_87 --gpu-code=sm_87 SMS=87 OPENMP=yes CUDA_PATH=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/
/usr/local/cuda-12.6 HOST_COMPILER=aarch64-yoe-linux-gcc  -march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-se
curity --sysroot=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot CCFLAGS=-march=armv8-a+crc -mbranch-protection=standard -fstack-protector-strong -O2 -D_FOR
TIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot  -O2 -g   -fmacro-prefix-map=/mnt
/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5  -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-sample
s/12.5/git=/usr/src/debug/cuda-samples/12.5  -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5  -fdebug-prefix-map=/m
nt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5  -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samp
les/12.5/recipe-sysroot=  -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot=  -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_te
gra234-yoe-linux/cuda-samples/12.5/recipe-sysroot-native=  -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot-native=  -pipe -I=/usr/local/c
uda-12.6/include LDFLAGS=-L/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/lib  --sysroot=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra2
34-yoe-linux/cuda-samples/12.5/recipe-sysroot --sysroot=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot -O1 --hash-style=gnu --as-needed -z,relro,-z,now -L=
/usr/local/cuda-12.6/lib -L=/usr/local/cuda-12.6/lib/stubs -rpath-link,/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/lib -rpath,/usr/l
ocal/cuda-12.6/lib -lstdc++ -lm TARGET_ARCH=aarch64 TARGET_OS=linux HOST_ARCH=x86_64 NVCC=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot-native//usr/local/
cuda-12.6/bin/nvcc -ccbin aarch64-yoe-linux-gcc-10.3.0 --include-path /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/include --library-
path /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/lib -Xcompiler -isystem,=/usr/include/cuda-compat-workarounds
| /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot-native//usr/local/cuda-12.6/bin/nvcc -ccbin aarch64-yoe-linux-gcc-10.3.0 --include-path /mnt/b/yoe/master/
build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/include --library-path /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe
-sysroot/usr/local/cuda-12.6/lib -Xcompiler -isystem,=/usr/include/cuda-compat-workarounds -I../../../Common -m64 -Xcompiler -march=armv8-a+crc -Xcompiler -mbranch-protection=standard -Xcompiler -fsta
ck-protector-strong -Xcompiler -O2 -Xcompiler -D_FORTIFY_SOURCE=2 -Xcompiler -Wformat -Xcompiler -Wformat-security -Xcompiler -Werror=format-security -Xcompiler --sysroot=/mnt/b/yoe/master/build/tmp/w
ork/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot -Xcompiler -O2 -Xcompiler -g -Xcompiler -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5 -Xcompiler -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5 -Xcompiler -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5 -Xcompiler -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/git=/usr/src/debug/cuda-samples/12.5 -Xcompiler -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot= -Xcompiler -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot= -Xcompiler -fdebug-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot-native= -Xcompiler -fmacro-prefix-map=/mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot-native= -Xcompiler -pipe -Xcompiler -I=/usr/local/cuda-12.6/include --threads 0 --std=c++11 --gpu-architecture=compute_87 --gpu-code=sm_87 -o asyncAPI.o -c asyncAPI.cu
| /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/include/bits/mathcalls.h(70): error: exception specification is incompatible with that of previous function "cospi" (declared at line 2595 of /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/include/crt/math_functions.h)
|   extern double cospi (double __x) noexcept (true); extern double __cospi (double __x) noexcept (true);
|                                    ^
|
| /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/include/bits/mathcalls.h(72): error: exception specification is incompatible with that of previous function "sinpi" (declared at line 2550 of /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/include/crt/math_functions.h)
|   extern double sinpi (double __x) noexcept (true); extern double __sinpi (double __x) noexcept (true);
|                                    ^
|
| /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/include/bits/mathcalls.h(70): error: exception specification is incompatible with that of previous function "cospif" (declared at line 2617 of /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/include/crt/math_functions.h)
|   extern float cospif (float __x) noexcept (true); extern float __cospif (float __x) noexcept (true);
|                                   ^
|
| /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/include/bits/mathcalls.h(72): error: exception specification is incompatible with that of previous function "sinpif" (declared at line 2573 of /mnt/b/yoe/master/build/tmp/work/armv8a_tegra234-yoe-linux/cuda-samples/12.5/recipe-sysroot/usr/local/cuda-12.6/include/crt/math_functions.h)
|   extern float sinpif (float __x) noexcept (true); extern float __sinpif (float __x) noexcept (true);
|                                   ^
|
| 4 errors detected in the compilation of "asyncAPI.cu".
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'master' with `p3737-0000-p3701-0005` based on 'agx'
2. Build with bitbake argument 'cuda-samples'
3. See error 'above' 

**Additional context**
[cuda-samples](https://github.com/NVIDIA/cuda-samples) repo has not tagged 12.6 release that maybe the issue.

madisongh, 2024-12-09:
The last build over the weekend [here](https://builder.madison.systems/#/builders/12/builds/28) was OK, so I don't think it's a version sync problem between the samples and CUDA. I don't see anything that's gone into OE-Core since Saturday that would cause this, either. Maybe a clang/LLVM conflict?

kraj, 2024-12-09:
> The last build over the weekend [here](https://builder.madison.systems/#/builders/12/builds/28) was OK, so I don't think it's a version sync problem between the samples and CUDA. I don't see anything that's gone into OE-Core since Saturday that would cause this, either. Maybe a clang/LLVM conflict?

Good point, although its not clang/llvm since its using gcc to compile this stuff, but I do have master branch of glibc being used and [recently](https://sourceware.org/git/?p=glibc.git;a=commit;h=3a9b4b4aeb7d7d9a26166b6b23a9fd762e1ffaa4), glibc has added C23 support which is adding some these functions as of today so maybe I will try with latest glibc today and see if that fixes the issue.

madisongh, 2024-12-09:
OK. If it's a glibc conflict, we may need to update the workaround I added with 981245c8bf3164f16bd7f52fca6ea9c9fa8c349a to deal with an earlier mismatch between the older stuff that nvcc expects and glibc's latest.

closed 2024-12-17 by madisongh