OE4T/meta-tegra #1787: Unable to Build Required Kernel Modules for Container Management on Jetson Nano with Yocto

JohnathanGamal, 2024-12-12:
## Issue: Kernel Modules Missing for Container-Management Feature in Custom Jetson Nano Yocto Image

### Problem:
I am building a custom Linux image for the Jetson Nano using Yocto with the `linux-tegra` kernel. During the build process, I encounter the following error:

[image]

The error is related to the missing kernel modules required for the container-management feature:
- `kernel-module-dm-thin-pool` (for Device Mapper thin provisioning)
- `kernel-module-xt-masquerade` (for Netfilter NAT functionality)

- **Yocto Layers in Use:**
`/layers/meta /layers/meta-poky /layers/meta-leda /layers/meta-leda/meta-leda-bsp /layers/meta-leda/meta-leda-components /layers/meta-leda/meta-leda-distro /layers/meta-rauc /layers/meta-kanto /layers/meta-virtualization /layers/meta-oe /layers/meta-filesystems /layers/meta-python /layers/meta-networking /layers/meta-tegra
`

- **Kernel Configuration:**
The custom kernel configuration for `linux-tegra` (located at `meta-tegra/recipes-kernel/linux/linux-tegra-4.9/custom-defconfig`) includes:

```
CONFIG_DM_THIN_PROVISIONING=m 
CONFIG_IP_NF_TARGET_MASQUERADE=m
```

These configurations should enable the required modules as loadable kernel modules (`m`).

- **Container-Management Recipe:**
The recipe for `container-management` (found in `meta-kanto/recipes-containers/container-management/container-management_git.bb`) has the following dependency:
`RDEPENDS:${PN} += "kernel-module-dm-thin-pool kernel-module-xt-masquerade"
`

### Expected Behavior:
The kernel modules `dm-thin-pool` and `xt-masquerade` should be included in the image, and the build should succeed without errors related to missing kernel modules.

### Current Behavior:
The build fails with errors about missing kernel modules (`kernel-module-dm-thin-pool` and `kernel-module-xt-masquerade`) required by the `container-management` package.

Any guidance on resolving this issue would be greatly appreciated.

### Environment:
- Yocto version: [insert Yocto version]
- Jetson Nano
- Custom Linux kernel (linux-tegra)

madisongh, 2024-12-12:
Start with checking to see if your kernel config modifications actually worked. You can examine the generated `.config` in the kernel build tree. If you don't see the settings the way you want them, examine the configuration logs to see why they didn't take.

If the settings are `=y` in the generated config, instead of `=m`, then the drivers are built into the kernel, rather than being separate modules, and the `RDEPENDS` in that recipe needs fixing - kernel module needs are usually set with `RRECOMMENDS` instead of `RDEPENDS` to allow for the possibility that needed drivers are built-in.

If the settings are `=m`, then packages for those drivers should have been generated, and should show up under `tmp/deploy/rpm` (assuming you're using the default RPM packaging) to available to the image recipe. You might want to look into the kernel makefiles to make sure that the driver names corresponding to those config setting actually match up with what the recipe is expecting.  The kernel for the TX1/TX2 is really old now, so there's a chance that drivers got renamed in later versions of the kernel and the recipe is trying to use the new names.

ichergui, 2025-01-20:
Hi @JohnathanGamal 
Any update on this issue ?

JohnathanGamal, 2025-01-20:
The issue was fixed by importing an older module instead of xt-masquerade to be compatible with the kernel version I am using.

closed 2025-01-20 by JohnathanGamal