OE4T/meta-tegra #1793: Default signing needs to pass CHIP_SKU to doflash, otherwise it fails on the new CHIP_SKU checking

eg9152, 2024-12-19:
In image_types_tegra.bbclass, tegraflash_custom_sign_pkg(), it is passing TEGRA_SIGNING_ENV to doflash.sh.  This is defined in tegra-common.inc and is missing CHIP_SKU.  It causes the CHIP_SKU check to fail when signing ("ERR: no default chip SKU set").

It looks like it was just missed when adding the new CHIP_SKU checking.

ichergui, 2024-12-19:
For offline singing make sure you are using the `RAMCODE`  and `CHIP_SKU`

madisongh, 2024-12-19:
If you're using one of the 36.4-based branches, the chip SKU should be getting passed through the flashvars - the TEGRA_FLASHVAR_CHIP_SKU variable should be set in the machine configuration.

If you have a custom machine configuration, it may need updating for this.

eg9152, 2024-12-20:
This is scarthgap, L4T 36.4.  I had missed that it should be getting it from flashvars.

I see what happened now.  The custom machine I'm using was started from a copy of p3768-0000-p3767-0000.conf, before there was an orin-nx.inc available.  In the latest updates, I had missed that the new variable TEGRA_FLASHVAR_CHIP_SKU was added to orin-nx.inc.  The flashvars recipe will silently not include a variable if it's not defined.  So my flashvars didn't have the CHIP_SKU line.

Thank you for your help!

closed 2024-12-20 by eg9152