OE4T/meta-tegra #1798: Hash of public_sources seems wrong

minicarpet, 2025-01-10:
**Describe the bug**
When starting yocto from scratch using branch scarthgap, I got the following error:
````
WARNING: nvidia-kernel-oot-36.4.0-r0 do_fetch: Checksum failure encountered with download of https://developer.download.nvidia.com/embedded/L4T/r36_Release_v4.0/sources/public_sources.tbz2;downloadfilename=Jetson-public_sources-36.4.0.tbz2 - will attempt other sources if available
WARNING: nvidia-kernel-oot-36.4.0-r0 do_fetch: Checksum mismatch for local file /home/yocto_user/build/downloads/Jetson-public_sources-36.4.0.tbz2
Cleaning and trying again.
WARNING: nvidia-kernel-oot-36.4.0-r0 do_fetch: Renaming /home/yocto_user/build/downloads/Jetson-public_sources-36.4.0.tbz2 to /home/yocto_user/build/downloads/Jetson-public_sources-36.4.0.tbz2_bad-checksum_5e0f936c52cb1e65ccda6fe580472f66166fa4687aeb931c2f0b25bf8c858daf
ERROR: nvidia-kernel-oot-36.4.0-r0 do_fetch: Checksum failure fetching https://developer.download.nvidia.com/embedded/L4T/r36_Release_v4.0/sources/public_sources.tbz2;downloadfilename=Jetson-public_sources-36.4.0.tbz2
ERROR: nvidia-kernel-oot-36.4.0-r0 do_fetch: Bitbake Fetcher Error: ChecksumError('Checksum mismatch!\nFile: \'/home/yocto_user/build/downloads/Jetson-public_sources-36.4.0.tbz2\' has sha256 checksum \'5e0f936c52cb1e65ccda6fe580472f66166fa4687aeb931c2f0b25bf8c858daf\' when \'a99cb9176a96cff19ddeb928c9e4a2c5d5bf45787b2c2941fd3e942ee7540292\' was expected\nIf this change is expected (e.g. you have upgraded to a new version without updating the checksums) then you can use these lines within the recipe:\nSRC_URI[sha256sum] = "5e0f936c52cb1e65ccda6fe580472f66166fa4687aeb931c2f0b25bf8c858daf"\nOtherwise you should retry the download and/or check with upstream to determine if the file has become corrupted or otherwise unexpectedly modified.', 'https://developer.download.nvidia.com/embedded/L4T/r36_Release_v4.0/sources/public_sources.tbz2;downloadfilename=Jetson-public_sources-36.4.0.tbz2')
ERROR: Logfile of failure stored in: /home/yocto_user/build/tmp/work/orin_devkit-poky-linux/nvidia-kernel-oot/36.4.0/temp/log.do_fetch.56143
ERROR: Task (/home/yocto_user/build/../layers/meta-tegra/recipes-kernel/nvidia-kernel-oot/nvidia-kernel-oot_36.4.0.bb:do_fetch) failed with exit code '1'
````

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'scarthgap' with `MACHINE` based on 'p3768-0000-p3767-0000'
2. Build with bitbake argument 'nvidia-kernel-oot'
4. See error 'wrong checksum'

**Additional context**
The strange behavior is when I download manually the file, I've got again a different checksum : 
````
yocto_user@f6757a551db1:~/build$ wget https://developer.download.nvidia.com/embedded/L4T/r36_Release_v4.0/sources/public_sources.tbz2
--2025-01-10 15:07:03--  https://developer.download.nvidia.com/embedded/L4T/r36_Release_v4.0/sources/public_sources.tbz2
Resolving developer.download.nvidia.com (developer.download.nvidia.com)... 96.16.248.136, 96.16.248.138
Connecting to developer.download.nvidia.com (developer.download.nvidia.com)|96.16.248.136|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 225930643 (215M) [text/plain]
Saving to: ‘public_sources.tbz2’

public_sources.tbz2                                  100%[===================================================================================================================>] 215.46M  60.7MB/s    in 4.0s

2025-01-10 15:07:07 (53.4 MB/s) - ‘public_sources.tbz2’ saved [225930643/225930643]

yocto_user@f6757a551db1:~/build$ sha256sum public_sources.tbz2
6e95abdf9195052b8d6b63777b9b1417a758248b2cf1a82c469ab75d2c4af754  public_sources.tbz2
````

I probably missunderstood something here

madisongh, 2025-01-10:
No, it looks like NVIDIA has once again done a rug pull on us, updating the tarball without bumping the version number.  The actual source files contained in the tarball are probably unchanged, so just updating the `tegra-sources-36.4.0.inc` file with the new sha256sum should work OK.

dwalkes, 2025-01-10:
Added a PR at https://github.com/OE4T/meta-tegra/pull/1799 in case this is useful, if it's easier to fix yourself @madisongh please ignore.

minicarpet, 2025-01-10:
That is sad that nVidia do this...
But just one question, why do yocto détective a different checksum while sha256sum has an other ? (See yocto log)

dwalkes, 2025-01-10:
> But just one question, why do yocto détective a different checksum while sha256sum has an other ? (See yocto log)

Not sure about this either, I was wondering if it changed twice?  I can confirm the second one you shared match what I saw, and the warning I get for my local file after changing was

```
WARNING: optee-os-tadevkit-4.2-l4t-r36.4.0-r0 do_fetch: Checksum mismatch for local file /home/dan/trellis/oe4t/tegra-demo-distro/build-master/downloads/Jetson-public_sources-36.4.0.tbz2
Cleaning and trying again.
WARNING: optee-os-tadevkit-4.2-l4t-r36.4.0-r0 do_fetch: Renaming /home/dan/trellis/oe4t/tegra-demo-distro/build-master/downloads/Jetson-public_sources-36.4.0.tbz2 to /home/dan/trellis/oe4t/tegra-demo-distro/build-master/downloads/Jetson-public_sources-36.4.0.tbz2_bad-checksum_a99cb9176a96cff19ddeb928c9e4a2c5d5bf45787b2c2941fd3e942ee7540292
```
So I never saw the `5e0f936c52cb1e65ccda6fe580472f66166fa4687aeb931c2f0b25bf8c858daf` sha256 sum myself.

minicarpet, 2025-01-10:
Ok I will try again then once PR is done

dwalkes, 2025-01-10:
You could also try ( on the meta-tegra layer)
```
git remote add trellis-logic https://github.com/Trellis-Logic/meta-tegra.git
git fetch trellis-logic
git checkout trellis-logic/fix-r36-hash
```
if you want to make sure it works for you as well before the PR merges.

closed 2025-01-10 by madisongh

alonbl, 2025-01-16:
Hi,
If nvidia modifies in-place files, we must host/mirror these nvidia versions ourselves, every time this change happens there are way too many changes go into the distribution.
What do you think?
Instead of downloading from nvidia, manage a git repo with the binaries in properly versioned.
Thanks,
Alon

alonbl, 2025-01-16:
> Hi, If nvidia modifies in-place files, we must host/mirror these nvidia versions ourselves, every time this change happens there are way too many changes go into the distribution. What do you think? [...]

I created a new issue for this: https://github.com/OE4T/meta-tegra/issues/1806