OE4T/meta-tegra #1803: Failed to validate rootfs status: Load Error

joeced, 2025-01-15:
**Describe the bug**
In an evaluation of Orin NX 16 GB, we are building an image akin to meta-tegra-demodistro,
with our own customizations. Notable, we change the the `L4TDefaultBootMode` to `02 00 00 00`:
(Boot normal kernel and DTB in partitions).

Using the [latest commit](https://github.com/OE4T/meta-tegra/commit/c7e77d1b8ef7377b8fa9064b1c800f0c354db927) on the scarthgap branch, we get an error on boot.

```
Failed to validate rootfs status: Load Error
```

Using the layers listed below (all on scarthgap), it works:
* meta-openembedded - 3c293e
* meta-swupdate - 4a65b1
* meta-tegra - 803485
* poky - 7117d1

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'scarthcap' with `MACHINE` based on 'orin nx 16 gb' with commit `c7e77d`
2. Build with `bitbake <our-image>`
3. Deploy to hardware with method 'tegraflash' with `./initrd-flash`
4. See error below

```
Jetson System firmware version v36.4.0 date 2024-10-01T15:28:28+00:00
ESC   to enter Setup.
F11   to enter Boot Manager Menu.
Enter to continue boot.
......
      Failed to validate rootfs status: Load Error
L4TLauncher: Attempting GRUB Boot
L4TLauncher: Attempting Direct Boot
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services...
�ebugfs initialized

... <lots of lines :)>

nv-devkit-orin-nx16 login:
nv-devkit-orin-nx16 login: root
root@nv-devkit-orin-nx16:~# nvbootctrl -t rootfs dump-slots-info
Error: read variable RootfsRedundancyLevel failed!
Error: invalid num_slots: return(-2)
root@nv-devkit-orin-nx16:~# 
```

**Additional context**
N/A

madisongh, 2025-01-16:
How are you changing `L4TDefaultBootMode`? Are you using a boot control DTB overlay for that, or something else?

Do you have any other customizations that would affect the EFI variables in some way?  The error messages are due to some EFI variables either being not set at all, or set to invalid values.

It's certainly possible that I broke something in the initrd-flash changes, and answers to the above could help me zero in on what to look for.

joeced, 2025-01-16:
I'm changing the `L4TDefaultBootMode` using the same method as shown in `meta-tegra/recipes-bsp/uefi/edk2-firmware-tegra_36.4.0.bb` with the `L4TConfiguration-RootfsRedundancyLevelABEnable.dtsi` file.

My input is this `dtsi` file:

```
/dts-v1/;

/ {
  overlay-name = "L4T Configuration Settings";

  fragment@0 {
    target-path = "/";

    board_config {
      sw-modules = "uefi";
    };
    __overlay__ {

      firmware {

        uefi {

          variables {

            gNVIDIAPublicVariableGuid {

              L4TDefaultBootMode {

                data = [02 00 00 00];
                runtime;
                non-volatile;
              };
            };
          };
        };
      };
    };
  };
};

```

madisongh, 2025-01-19:
I just merged some updates to the flashing scripts that should fix this.

closed 2025-01-19 by madisongh

joeced, 2025-01-21:
Confirmed! It worked 👍

PR #1815 (2025-01-19): More fixes for getting initrd flashing to work with pre-signed binaries
* Fixed a typo that would omit some DTB overlays for some cases
* Reworked the use of the RCM boot blob signing in the `odmsign.func` functions so the signed files for RCM booting don't end up replacing the main signed files
* Signing step in initrd-flash now skips signing partitions on external storage devices
* Since booting using pre-signed boot firmware prevents us from reading the CVM EEPROM to get the serial number, added a mechanism for locating the right storage device based on the USB instance being in the device  [... cut]