OE4T/meta-tegra #1806: Host nvidia artifacts in meta-tegra artifact repository since nvidia modifies files in-place at their site

alonbl, 2025-01-16:
Hello,

nvidia does not understand what is reproducible build nor what release term is, they modify the released artifacts without allowing to keep up with older releases or verify the changes that are going into the image.

At least the following events:

* https://github.com/OE4T/meta-tegra/pull/1799
* https://github.com/OE4T/meta-tegra/issues/1798
* https://github.com/OE4T/meta-tegra/issues/237

The problems:

1. User is forced to sync with the OE4T branch getting more changes than needed only to accept the hash change patch.
2. User is forced to apply changes from the nvidia modifications as something is modified, even if it is "harmless".
3. User is unable to perform regression because he cannot build older tagged images as files are incorrect and not available.

Please consider opening a new git repo and host all nvidia artifacts in this repo, so that we have control over what goes into our images.

Modifying hash in-place must not be done, the yocto layer cannot tolerate this.

Thanks,

alimon, 2025-01-17:
Hi, I ended up doing it with the Nvidia BSP tarball [1] too because if you are adding custom board support the same problem can appear and is better to track changes in a repository.

Of course there are maintainence efforts associated.

[1] https://developer.nvidia.com/downloads/embedded/l4t/r36_release_v4.0/release/Jetson_Linux_R36.4.0_aarch64.tbz2

madisongh, 2025-01-17:
Besides the maintenance efforts and costs involved, I'm not sure licensing would permit redistribution of that sort, at least for some of the artifacts.  You can maintain your own downloads cache for your own use - that's generally good practice.

It's definitely annoying that we had another one of these release engineering newbie mistakes happen recently. NVIDIA has generally been better about this stuff than they were in the past. I'm not sure putting the onus on this unfunded, all-volunteer project to compensate for those mistakes is necessarily the right answer, though.

closed 2025-01-17 by madisongh

alonbl, 2025-01-17:
As far as licensing are concerned once there is a public URL there is no restriction of mirror.

Maintaining my own download cache does not work, you cannot go forward/backward with same artifact names with different hash, the artifact name must be modified in the mirror and match in the tree to make it work.

I am sure that all-volunteer project that knows much more about stability than nvidia will be able to compensate a company that releases artifacts without versions and have no culture of stability (EVER!).

It will require less effort that chasing breakage every time someone at nvidia releases another version, it will require less effort for the users of yocto to recover from the situation.

It is very sad that you close it, are you open to discuss how to provide an alternate method to solve this? for example, have a centralized include file for nvidia artifacts so people may establish such a mirror using an override?

Do you know anyone at nvidia we can talk with to explain what release engineering is in the real world?

madisongh, 2025-01-19:
We have had discussions with NVIDIA in the past about release engineering topics - see [this wiki page](https://github.com/OE4T/meta-tegra/wiki/ZZZArchived:-L4T-Integration-Issues) for some of them. There have also been postings to their developer forum when other issues came up. To their credit, they took many of these comments on board, and as I said earlier, the situation has definitely improved over the early days.

If someone wants to embark on a project to provide a public mirror of the NVIDIA downloads, and needs some changes in the layer here to simplify the use of that mirror, I'm fine with discussing such changes... either in a [discussion thread](https://github.com/OE4T/meta-tegra/discussions), or over a pull request once the specific changes have been worked out.