OE4T/meta-tegra #1820: Unable to flash only the device tree using doflash.sh

fdiazRR, 2025-01-24:
Flashing only the device tree using doflash fails to find the partition type for the DTB

Whenever I try to flash only the device tree using the following command
`sudo ./doflash.sh -c "sign; write DTB signed/tegra234-p3737-0000+p3701-0005-nv.dtb.signed"`

I run into the following error
`[   7.9200 ] Sending membct and RCM blob`
`[   7.9203 ] tegrarcm_v2 --chip 0x23 0 --pollbl --download bct_mem mem_rcm_sigheader.bct.encrypt --download blob blob.bin`
`[   7.9205 ] BL: version 1.4.0.2-t234-54845784-08a4de08 last_boot_error: 0`
`[   7.9570 ] Sending bct_mem`
`[   8.0093 ] Sending blob`
`[   9.9121 ] completed`
`[  10.9159 ] tegrarcm_v2 --chip 0x23 0 --ismb2applet`
`[  11.2547 ] tegrarcm_v2 --chip 0x23 0 --ismb2`
`[  11.2554 ] MB2 version 01.00.0000`
`Error: Error: Can not find partition type for dtb`

If I run a complete flash of the board, the doflash.sh is able to find the partition type for the device tree and flash all the partitions of the board

**To Reproduce**
Build meta-tegra normally and flash the board.
Then modify the device tree (or just change its name) and try to flash only the device tree with the command given above

madisongh, 2025-01-25:
You don't say which branch you're using, but it looks like it's one of the branches with L4T R36.4, based on the name of the DTB file.

The `write` command you are passing to tegraflash takes the name of a partition to write. Check your storage layout XML file to see the partition names, but the ones for the kernel device tree are probably called `A_kernel-dtb` and `B_kernel-dtb`, not `DTB`. Using the name of an existing partition should work better.

That said, if you need to test device tree changes, it's probably easier to use the `UBOOT_EXTLINUX_FDT` variable to have your DTB file load from `/boot`.  See discussion #1670  for more information... assuming you are still using the default L4TLauncher to load the kernel/devicetree/etc.

fdiazRR, 2025-01-29:
We've used the `UBOOT_EXTLINUX_FDT` approach before and that's what we commonly do but due to the project requirements we needed a way of flashing directly the device tree partition.

After specifying `A_kernel-dtb` and `B_kernel-dtb` I was able to flash the device tree partition. Thank you for the support.

closed 2025-01-29 by madisongh