OE4T/meta-tegra #1824: initrd_flash fails

Lander0606, 2025-01-29:
When executing the _initrd_flash_ tool, I get this error:
```
lander@fedora:~/tegraflash$ sudo ./initrd-flash
Starting at 2025-01-29T22:07:35+01:00
Machine:       jetson-orin-nano-devkit
Rootfs device: mmcblk0p1
Found Jetson device in recovery mode at USB 1-6
== Step 1: Signing binaries at 2025-01-29T22:07:35+01:00 ==
ERR: chip_info.bin_bak missing after dumping boardinfo
```

If I try to use doflash, I get the same error about the missing _chip_info.bin_bak_. The Jetson Orin Nano is booted in recovery mode, uses the newest firmware from NVIDIA (support up to Jetpack 6.2) and I used meta-tegra branch "scarthgap" to generate my core-image-base. Any idea how to solve this?

madisongh, 2025-01-29:
Were there any error messages in the `log.initrd-flash.XXXXXXX` file, or when you use doflash.sh, do you get any error messages besides the final one about the missing file?

Lander0606, 2025-01-29:
I can see these messages in the log:
```
[   0.0135 ] tegrarcm_v2 --instance 1-6 --chip 0x23 0 --ismb2applet
[   0.0137 ] File rcm_state open failed
[   0.0139 ] ERROR: failed to read rcm_state
```
```
[   0.1171 ] Sending bct_br
[   0.1213 ] ERROR: might be timeout in USB write.
Error: Return value 3
Command tegrarcm_v2 --instance 1-6 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader.bin.encrypt --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader.bin.encrypt --download bct_mb1 mb1_bct_MB1_sigheader.bct.encrypt
```
```
[  10.5955 ] tegrabct_v2 --brbct tmp.bct --chip 0x23 0 --custinfo /home/lander/tegraflash/custinfo_out.bin
[  10.5956 ] File tmp.bct open failed
[  10.5957 ] Error: try getting custinfo fail. Moving on
[  10.5957 ] Rebooting to recovery mode
[  10.5959 ] tegrarcm_v2 --instance 1-6 --chip 0x23 0 --ismb2
[  10.5961 ] Rebooting to recovery mode
[  10.5963 ] tegrarcm_v2 --instance 1-6 --chip 0x23 0 --reboot recovery
ERR: signing failed at 2025-01-29T22:07:47+01:00
```

From what I can see is the board booted in recovery mode, using a jumper on the pins like described in the documentation.

Lander0606, 2025-01-30:
The issue seems to be with Boxes in Linux, because even the SDK Manager failed to flash the device. Using a Fedora 41 host, the flashing was successful.

closed 2025-01-30 by Lander0606

madisongh, 2025-01-30:
Yes, it's well-known that the NVIDIA flashing tools don't typically work from a VM.  I've added a note about that to our [wiki page](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit).