OE4T/meta-tegra #1827: Errors building an image for jetson-xavier-nx-devkit-emmc on kirkstone and scarthgap-l4t-r35.x

jdiez17, 2025-01-30:
**Describe the bug**

I get the following errors trying to build an image using the meta-tegra layer:

```
ERROR: tegra-nvpmodel-35.6.0-r0 do_package: QA Issue: tegra-nvpmodel: Files/directories were installed but not shipped in any package:
  /usr
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tegra-nvpmodel: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: tegra-redundant-boot-35.6.0-r1 do_package: QA Issue: tegra-redundant-boot: Files/directories were installed but not shipped in any package:
  /usr
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tegra-redundant-boot: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: tegra-nvs-service-35.6.0-r0 do_package: QA Issue: tegra-nvs-service: Files/directories were installed but not shipped in any package:
  /usr
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tegra-nvs-service: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: tegra-nvstartup-35.6.0-r0 do_package: QA Issue: tegra-nvstartup: Files/directories were installed but not shipped in any package:
  /usr
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tegra-nvstartup: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: tegra-nvphs-35.6.0-r0 do_package: QA Issue: tegra-nvphs: Files/directories were installed but not shipped in any package:
  /usr
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tegra-nvphs: 1 installed and not shipped files. [installed-vs-shipped]
ERROR: tegra-nvsciipc-35.6.0-r0 do_package: QA Issue: tegra-nvsciipc: Files/directories were installed but not shipped in any package:
  /usr
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
tegra-nvsciipc: 1 installed and not shipped files. [installed-vs-shipped]
```

This happens on the `kirkstone` branch and `scarthgap`.

ichergui, 2025-01-30:
Hi @jdiez17 
Thanks for reporting this issue.
I will check that and will fix it.

jdiez17, 2025-01-30:
@ichergui Thank you for the very fast response! I was able to build on kirkstone with the following patch:

```diff
diff --git a/recipes-bsp/tegra-binaries/tegra-nvphs_35.6.0.bb b/recipes-bsp/tegra-binaries/tegra-nvphs_35.6.0.bb
index 4f74fd84..ee8e86b2 100644
--- a/recipes-bsp/tegra-binaries/tegra-nvphs_35.6.0.bb
+++ b/recipes-bsp/tegra-binaries/tegra-nvphs_35.6.0.bb
@@ -21,6 +21,7 @@ do_install() {

 inherit systemd update-rc.d

+INSANE_SKIP:${PN} = "installed-vs-shipped"
 INITSCRIPT_NAME = "nvphs"
 INITSCRIPT_PARAMS = "defaults"
 SYSTEMD_SERVICE:${PN} = "nvphs.service"
diff --git a/recipes-bsp/tegra-binaries/tegra-nvpmodel_35.6.0.bb b/recipes-bsp/tegra-binaries/tegra-nvpmodel_35.6.0.bb
index bbd76acf..e60f6072 100644
--- a/recipes-bsp/tegra-binaries/tegra-nvpmodel_35.6.0.bb
+++ b/recipes-bsp/tegra-binaries/tegra-nvpmodel_35.6.0.bb
@@ -21,7 +21,9 @@ do_install() {

 inherit systemd update-rc.d

+
+INSANE_SKIP:${PN} = "installed-vs-shipped"
 INITSCRIPT_NAME = "nvpmodel"
 INITSCRIPT_PARAMS = "defaults"
 SYSTEMD_SERVICE:${PN} = "nvpmodel.service"
-RDEPENDS:${PN} = "tegra-nvpmodel-base tegra-nvpower"
+RDEPENDS:${PN} = "tegra-nvpmodel-base tegra-nvpower"
\ No newline at end of file
diff --git a/recipes-bsp/tegra-binaries/tegra-nvs-service_35.6.0.bb b/recipes-bsp/tegra-binaries/tegra-nvs-service_35.6.0.bb
index 23e0028f..0db9f0d7 100644
--- a/recipes-bsp/tegra-binaries/tegra-nvs-service_35.6.0.bb
+++ b/recipes-bsp/tegra-binaries/tegra-nvs-service_35.6.0.bb
@@ -21,6 +21,7 @@ do_install() {

 inherit systemd update-rc.d

+INSANE_SKIP:${PN} = "installed-vs-shipped"
 INITSCRIPT_NAME = "nvs-service"
 INITSCRIPT_PARAMS = "defaults"
 SYSTEMD_SERVICE:${PN} = "nvs-service.service"
diff --git a/recipes-bsp/tegra-binaries/tegra-nvsciipc_35.6.0.bb b/recipes-bsp/tegra-binaries/tegra-nvsciipc_35.6.0.bb
index 7d326694..7687c9e5 100644
--- a/recipes-bsp/tegra-binaries/tegra-nvsciipc_35.6.0.bb
+++ b/recipes-bsp/tegra-binaries/tegra-nvsciipc_35.6.0.bb
@@ -29,3 +29,4 @@ INITSCRIPT_PARAMS = "defaults"
 SYSTEMD_SERVICE:${PN} = "nv_nvsciipc_init.service"

 RDEPENDS:${PN} = "tegra-nvsciipc-base"
+INSANE_SKIP:${PN} = "installed-vs-shipped"
\ No newline at end of file
diff --git a/recipes-bsp/tegra-binaries/tegra-nvstartup_35.6.0.bb b/recipes-bsp/tegra-binaries/tegra-nvstartup_35.6.0.bb
index 509ed0e3..05515b34 100644
--- a/recipes-bsp/tegra-binaries/tegra-nvstartup_35.6.0.bb
+++ b/recipes-bsp/tegra-binaries/tegra-nvstartup_35.6.0.bb
@@ -20,6 +20,7 @@ do_install() {

 inherit systemd update-rc.d

+INSANE_SKIP:${PN} = "installed-vs-shipped"
 INITSCRIPT_NAME = "nvstartup"
 INITSCRIPT_PARAMS = "defaults 00"
 SYSTEMD_SERVICE:${PN} = "nvstartup.service"
diff --git a/recipes-bsp/tegra-binaries/tegra-redundant-boot_35.6.0.bb b/recipes-bsp/tegra-binaries/tegra-redundant-boot_35.6.0.bb
index bf09938e..d669c1b5 100644
--- a/recipes-bsp/tegra-binaries/tegra-redundant-boot_35.6.0.bb
+++ b/recipes-bsp/tegra-binaries/tegra-redundant-boot_35.6.0.bb
@@ -22,6 +22,7 @@ do_install() {
[... 14 lines cut ...]
DEBUG: Fetcher accessed the network with the command LANG=C git -c core.fsyncobjectfiles=0 -c gc.autoDetach=false -c core.pager=cat clone --bare --mirror https://github.com/OE4T/linux-tegra-5.10.git /home/builder/raccoon-os/build/downloads/git2/github.com.OE4T.linux-tegra-5.10.git --progress
DEBUG: Running export PSEUDO_DISABLED=1; export PATH="/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/usr/bin/kern-tools-tegra:/home/builder/raccoon-os/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/usr/bin/python3-native:/home/builder/raccoon-os/src/poky/scripts:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot/usr/bin/crossscripts:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/usr/sbin:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/usr/bin:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/sbin:/home/builder/raccoon-os/build/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/linux-tegra/5.10.216+git-r0/recipe-sysroot-native/bin:/home/builder/raccoon-os/src/poky/bitbake/bin:/home/builder/raccoon-os/build/tmp/hosttools"; export HOME="/tmp/tmpgbkily5u"; LANG=C git -c core.fsyncobjectfiles=0 -c gc.autoDetach=false -c core.pager=cat clone --bare --mirror https://github.com/OE4T/linux-tegra-5.10.git /home/builder/raccoon-os/build/downloads/git2/github.com.OE4T.linux-tegra-5.10.git --progress
Cloning into bare repository '/home/builder/raccoon-os/build/downloads/git2/github.com.OE4T.linux-tegra-5.10.git'...
/home/builder/raccoon-os/build/downloads/git2/github.com.OE4T.linux-tegra-5.10.git/branches/: No such file or directory
```

Very strange. The workaround I used to fix it is also unusual: 

```diff
diff --git a/recipes-kernel/linux/linux-tegra_5.10.bb b/recipes-kernel/linux/linux-tegra_5.10.bb
index e787ee64..fab81578 100644
--- a/recipes-kernel/linux/linux-tegra_5.10.bb
[... cut ...]
 SRCREV = "4bce4d148ef3ff159ab55c127d8761aeaac5cc28"
 KBRANCH = "${SRCBRANCH}"
-SRC_REPO = "github.com/OE4T/linux-tegra-5.10.git;protocol=https"
+SRC_REPO = "github.com/OE4T/linux-tegra-5.10;protocol=https"
 KERNEL_REPO = "${SRC_REPO}"
 SRC_URI = "git://${KERNEL_REPO};name=machine;branch=${KBRANCH} \
            ${@'file://localversion_auto.cfg' if d.getVar('SCMVERSION') == 'y' else ''} \
```

Would be good to know why this is happening :) Or if it's just a problem on my setup. But I don't think so.

ichergui, 2025-01-30:
I started building `scarthgap-l4t-r35.x` branch with `demo-image-base` image and no issues from my side. 
I couldn't reproduce your issue.
which image your building ?

madisongh, 2025-02-08:
> Would be good to know why this is happening :) Or if it's just a problem on my setup. But I don't think so.

I think maybe it is your setup. You could compare yours with a build using our [demo distro](https://github.com/OE4T/tegra-demo-distro) setup to see what's different.

jdiez17, 2025-02-08:
Hi, sorry I haven’t had a chance to run more test builds to troubleshoot this. 

I suspect we may be using an older/different commit of poky or meta-oe in our setup. I will post a reproducer config next time I have a chance to run some more builds. To be continued…