OE4T/meta-tegra #1828: Jetson Orin Nano serial console errors

Lander0606, 2025-01-30:
I created an image for the Jetson Orin Nano (devkit) but in the serial console (debug) of the Jetson I'm stuck in a login loop. I get numerous amounts of login prompts until the maximum amount of login attempts is exceeded: 

```
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
Poky (Yocto Project Reference Distro) 5.0.6 jetson-orin-nano-devkit /dev/ttyTCUO
jetson-orin-nano-devkit login:
```

Also, when inserting an ethernet cable, I see no ip address being assigned to the board. Any idea how to solve these errors?

The firmware of the board is updated to version ```36.4.2``` be compatible with Jetpack 6.2 and the board is flashed with my image using ```initrd-flash```. No errors occurred while flashing.

Build target: ```core-image-base```
Yocto version: ```scarthgap```
Storage device: SD card with rootfs device ```mmcblk0p1```

madisongh, 2025-01-30:
Sounds like you're using stock poky... could you try building with `INIT_MANAGER = "systemd"`  in your configuration? The L4T stuff is all produced by NVIDIA for systemd-based distros, and while we try to retrofit them for compatibility with sysvinit, it doesn't get tested very frequently.

Alternatively, you could try using our [demo distro](https://github.com/OE4T/tegra-demo-distro) to get started. We build and test that regularly, using the `demo-image-*` recipes there.

Lander0606, 2025-01-30:
I added ```systemd``` to the DISTRO_FEATURES in my ```local.conf``` (adding it in core-image-base didn't change anything), but now I get the following error when building:

```
ERROR: Nothing RPROVIDES 'udev' (but /home/landervl/yocto-tegra/poky/meta/recipes-core/packagegroups/packagegroup-core-boot.bb RDEPENDS on or otherwise requires it)
eudev RPROVIDES udev but was skipped: conflicting distro feature 'systemd' (in DISTRO_FEATURES)
systemd RPROVIDES udev but was skipped: missing required distro feature 'usrmerge' (not in DISTRO_FEATURES)
```

I followed the tutorial from RidgeRun to create a custom Linux distribution for the Jetson boards, so I just cloned poky and meta-tegra in a folder with the following local.conf additions:

```
MACHINE ?= "jetson-orin-nano-devkit"

DISTRO_FEATURES = "x11 opengl systemd"

IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
```

I then chose the ```core-image-base``` build target, but I'm starting to think that this tutorial isn't up to date anymore or things changed for the Jetson Orin Nano...

Do I need some extra layers in Yocto or other changes in my local.conf? Or are there maybe other (new) documentation pages to consult? I'm going to make a Linux operating system for the Jetson Orin Nano in the context of a master's thesis "Linux OS in Space", so this will involve stripping the current distribution and customizing the kernel.

dwalkes, 2025-01-30:
> but I'm starting to think that this tutorial isn't up to date anymore

I do think the page at https://developer.ridgerun.com/wiki/index.php/Yocto_Support_for_NVIDIA_Jetson_Platforms_-_Setting_up_Yocto is possibly out of date, but they did add a link there to tegra-demo-distro which is what I'd recommend for getting started (as Matt suggested) along with `demo-image-base` or similar image.

Once you have a working image there you can strip things out and/or compare to the base poky setup more easily.

closed 2025-02-02 by Lander0606