OE4T/meta-tegra #1844: Root File System Redundancy is enabled with USE_REDUNDANT_FLASH_LAYOUT set to 0

michielskobe, 2025-02-20:
**Describe the bug**

I set `USE_REDUNDANT_FLASH_LAYOUT` to 0 in my `local.conf`, but when I build my image using the `dosdcard.sh` script I still have an extra rootfs partition `APP_b`.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch `master:0c07ff6ac8434afc391695bd3687355f7b4436f2` with `MACHINE` on `jetson-orin-nano-devkit`.
2. Build a target. (We used bitbake argument `spacelinux-image-base`, based on `core-image-based`. This is a target from my own distro that has proved to be working on the development board.)
3. Create an image with the `dosdcard.sh`script from `tegraflash`, located in `build/tmp/deploy/images/jetson-orin-nano-devkit/spacelinux-image-base-jetson-orin-nano-devkit.rootfs.tegraflash.tar.gz`.

**Additional context**
I used `bitbake -e` to troubleshoot some of the variable settings and got the following results:
```
USE_REDUNDANT_FLASH_LAYOUT_DEFAULT="0"
USE_REDUNDANT_FLASH_LAYOUT="0"

ROOTFSPART_SIZE_DEFAULT="30064771072"
ROOTFSPART_SIZE_REDUNDANT="15032385536"
ROOTFSPART_SIZE="30064771072"

PARTITION_LAYOUT_TEMPLATE_DEFAULT="flash_t234_qspi_sd.xml"
PARTITION_LAYOUT_TEMPLATE_REDUNDANT="flash_t234_qspi_sd_rootfs_ab.xml"
PARTITION_LAYOUT_TEMPLATE="flash_t234_qspi_sd.xml"
PARTITION_LAYOUT_TEMPLATE_DEFAULT_SUPPORTS_REDUNDANT="0"
```

The `internal-flash.xml` that is used to create the tegraflash package [here](https://github.com/OE4T/meta-tegra/blob/4d827691dd35c61f2f72f75ff07f0bb945d1713a/classes-recipe/image_types_tegra.bbclass#L177) does not contain a partition `APP_b` for the redundant rootfs.

After this, I took a look at `dosdcard.sh` itself. This used `flash.xml.in`, which did contain a partition for `APP_b`. I then created a log of the `dosdcard` script with the following command:

`sudo bash -x ./dosdcard.sh image.img | tee dosdcard.log`

When inspecting dosdcard.log, I noticed this at line 14:

`[   0.4536 ] Concatenating L4TConfiguration.dtbo,L4TConfiguration-RootfsRedundancyLevelABEnable.dtbo,tegra234-carveouts.dtbo,tegra-optee.dtbo,tegra234-p3768-0000+p3767-0000-dynamic.dtbo to tegra234-p3768-0000+p3767-0005-nv_with_odm_overlay.dtb.updated`

After some more looking around, I found that this `L4TConfiguration-RootfsRedundancyLevelABEnable.dtbo` is added [here](https://github.com/OE4T/meta-tegra/blob/4d827691dd35c61f2f72f75ff07f0bb945d1713a/conf/machine/include/tegra-common.inc#L20). Hence, it should only be added if USE_REDUNDANT_FLASH_LAYOUT is enabled, which is not the case so it should not be added.

madisongh, 2025-02-21:
> After some more looking around, I found that this L4TConfiguration-RootfsRedundancyLevelABEnable.dtbo is added [...]

Yes, exactly, it wouldn't. Can you clean your build tree and run a build from scratch, just to make sure you're really looking at the right output?

dwalkes, 2025-02-22:
@michielskobe I tried to reproduce with the latest on the master branch at https://github.com/OE4T/tegra-demo-distro and could not.

Here's what I did:
```
git clone https://github.com/OE4T/tegra-demo-distro
cd tegra-demo-distro
git checkout master
source setup-env --machine jetson-orin-nano-devkit
```

Then edit conf/local.conf to include this line:
```
USE_REDUNDANT_FLASH_LAYOUT = "0"
```

Then `bitbake demo-image-base`

Then extract the resulting tegraflash file in a new directory:
```
mkdir deploy
cd deploy
ln -s ../tmp/deploy/images/jetson-orin-nano-devkit/demo-image-base-jetson-orin-nano-devkit.rootfs.tegraflash.tar.zst
tar -xvf demo-image-base-jetson-orin-nano-devkit.rootfs.tegraflash.tar.zst
```

Then run `sudo ./doflash.sh`.  The output of tegraparser flash.xml parse looked like this and does not include `APP_b` partitions, as expected:
```
[  10.9796 ] tegraparser_v2 --storageinfo storage_info.bin --generategpt --pt flash.xml.bin
[  10.9801 ] gpt_secondary_3_0.bin:
[  10.9809 ] partition_id       partition_name                        StartingLba       EndingLba
[  10.9813 ]            1       BCT                                            0            2047
[  10.9818 ]            2       A_mb1                                       2048            3071
[  10.9822 ]            3       A_psc_bl1                                   3072            3583
[  10.9827 ]            4       A_MB1_BCT                                   3584            3839
[  10.9832 ]            5       A_MEM_BCT                                   3840            4351
[  10.9836 ]            6       A_tsec-fw                                   4352            6399
[  10.9841 ]            7       A_nvdec                                     6400            8447
[  10.9846 ]            8       A_mb2                                       8448            9471
[  10.9850 ]            9       A_xusb-fw                                   9472            9983
[  10.9855 ]           10       A_bpmp-fw                                   9984           13055
[  10.9860 ]           11       A_bpmp-fw-dtb                              13056           21247
[  10.9864 ]           12       A_psc-fw                                   21248           22783
[  10.9866 ]           13       A_mts-mce                                  22784           23807
[  10.9866 ]           14       A_sc7                                      23808           24191
[  10.9866 ]           15       A_pscrf                                    24192           24575
[  10.9866 ]           16       A_mb2rf                                    24576           24831
[  10.9866 ]           17       A_cpu-bootloader                           24832           31999
[  10.9866 ]           18       A_secure-os                                32000           40191
[  10.9866 ]           19       A_smm-fw                                   40192           44287
[  10.9866 ]           20       A_eks                                      44288           44799
[  10.9866 ]           21       A_dce-fw                                   44800           55039
[  10.9867 ]           22       A_spe-fw                                   55040           56191
[  10.9867 ]           23       A_rce-fw                                   56192           58239
[  10.9867 ]           24       A_adsp-fw                                  58240           62335
[  10.9867 ]           25       A_pva-fw                                   62336           62847
[  10.9867 ]           26       A_reserved_on_boot                         62848           65023
[  10.9867 ]           27       B_mb1                                      65024           66047
[  10.9867 ]           28       B_psc_bl1                                  66048           66559
[  10.9867 ]           29       B_MB1_BCT                                  66560           66815
[  10.9867 ]           30       B_MEM_BCT                                  66816           67327
[  10.9867 ]           31       B_tsec-fw                                  67328           69375
[  10.9867 ]           32       B_nvdec                                    69376           71423
[  10.9868 ]           33       B_mb2                                      71424           72447
[  10.9868 ]           34       B_xusb-fw                                  72448           72959
[  10.9868 ]           35       B_bpmp-fw                                  72960           76031
[  10.9868 ]           36       B_bpmp-fw-dtb                              76032           84223
[  10.9868 ]           37       B_psc-fw                                   84224           85759
[  10.9868 ]           38       B_mts-mce                                  85760           86783
[  10.9868 ]           39       B_sc7                                      86784           87167
[  10.9868 ]           40       B_pscrf                                    87168           87551
[... 61 lines cut ...]
```
[   0.3613 ] Concatenating L4TConfiguration.dtbo,tegra234-carveouts.dtbo,tegra-optee.dtbo,tegra234-p3768-0000+p3767-0000-dynamic.dtbo to tegra234-p3768-0000+p3767-0005-nv_with_odm_overlay.dtb.updated
```
and
````
Creating partitions
  [02] name=A_kernel start=0 size=262144 sectors
  [03] name=A_kernel-dtb start=0 size=1536 sectors
  [04] name=A_reserved_on_user start=0 size=64768 sectors
  [05] name=B_kernel start=0 size=262144 sectors
  [06] name=B_kernel-dtb start=0 size=1536 sectors
[... cut ...]
  [15] name=reserved start=0 size=982016 sectors
  [01] name=APP (fills to end)
```
So I think there's something different about your setup.  Can you compose a public minimal reproducible example based on tegra-demo-distro you could share?  Or could you compare tegra-demo-distro with yours to figure out where the problem is in yours?

Can you confirm you are deleting and recreating the tegraflash directory after rebuilding and not extracting over old contents or attempting to use previous tegraflash content?

michielskobe, 2025-03-18:
This response is a bit late, but this issue has been resolved for a bit. I an still not sure what was causing it, but cleaning everything and starting a fresh build did solve it.

closed 2025-03-18 by michielskobe