OE4T/meta-tegra #1852: RootfsRedundancyLevel is 0 when flashing A/B rootfs with initrd-flash

rambrandt, 2025-02-25:
**Describe the bug**

UEFI environment variable RootfsRedundancyLevel is 0 when flashing A/B rootfs with tegraflash/initrd-flash script, i.e. A/B rootfs is not enabled. It should be 1.

**How to reproduce**

Prerequisite for custom carrier: Patched `init-flash.sh` (the `/init` script called
by the kernel). It requires an additional kernel module and a wait-loop for UDC to be ready.

1.  Add external device (not necessary, turned out to happen also with eMMC) and build    Add `TNSPEC_BOOTDEV` to the image recipe and bitbake the image, MACHINE=jetson-agx-orin-carrier.

    TNSPEC_BOOTDEV:jetson-agx-orin-carrier="nvme0n1p1"

2. Unpack the resulting tegraflash archive and flash the image

    $ mkdir tegraflash-nvme && cd tegraflash-nvme
    $ tar xf .../my-image-dev-nvme-jetson-agx-orin-carrier.rootfs.tegraflash.tar.gz .
    $ sudo ./initrd-flash

   The script will erase the eMMC and install an A/B rootfs on /dev/nvme0n1.

3.  Check A/B UEFI variables

    root@jetson-agx-orin-carrier:~# nvbootctrl -t rootfs dump-slots-info
    RootFS A/B is not enabled.

    root@jetson-agx-orin-carrier:~# efivar -l | grep Rootfs
    781e084c-a330-417c-b678-38e696380cb9-RootfsStatusSlotB
    781e084c-a330-417c-b678-38e696380cb9-RootfsStatusSlotA
    781e084c-a330-417c-b678-38e696380cb9-RootfsRetryCountMax
    781e084c-a330-417c-b678-38e696380cb9-RootfsRedundancyLevel

    root@jetson-agx-orin-carrier:~# efivar -p -n 781e084c-a330-417c-b678-38e696380cb9-RootfsRedundancyLevel
    GUID: 781e084c-a330-417c-b678-38e696380cb9
    Name: "RootfsRedundancyLevel"
    Attributes:
            Boot Service Access
            Runtime Service Access
    Value:
    00000000  00 00 00 00                                       |....            |

    The zero value is contained in `L4TConfiguration-rcmboot.dtbo`.
   `L4TConfiguration-RootfsRedundancyLevelABEnable.dtbo` contains one.

    Tried the same without `TNSPEC_BOOTDEV`, i.e. with eMMC. Same result with
    `initrd-flash`. OK with `./doflash.sh`.
    Also tried with `initrd-flash --erase-nvme`.

    After successful AB flash with `./doflash.sh` I tried `./initrd-flash` again
    to check if it just misses something, but problem is back.

    The reason is that when `also_sign_rcmboot` is set (e.g. for eMMC), then
    `BOOTCONTROL_OVERLAYS` are not applied, but `rcm_bootcontrol_overlay` instead.

**Workaround**

 In `tegra-flash-helper.sh` replace in line 188

    rcm_bootcontrol_overlay="L4TConfiguration-rcmboot.dtbo"

 with:

    rcm_bootcontrol_overlay="${BOOTCONTROL_OVERLAYS:-L4TConfiguration-rcmboot.dtbo}"

rambrandt, 2025-02-25:
`USE_REDUNDANT_FLASH_LAYOUT_DEFAULT` was set.
```
    $ bitbake my-image-dev-nvme -e | grep ^USE_REDUNDANT_FLASH_LAYOUT
    USE_REDUNDANT_FLASH_LAYOUT="1"
    USE_REDUNDANT_FLASH_LAYOUT_DEFAULT="1"

    $ bitbake my-image-dev-nvme -e | grep ^TEGRA_BOOTCONTROL_OVERLAYS
    TEGRA_BOOTCONTROL_OVERLAYS="L4TConfiguration.dtbo  L4TConfiguration-RootfsRedundancyLevelABEnable.dtbo"
```

dwalkes, 2025-02-25:
> The reason is that when also_sign_rcmboot is set (e.g. for eMMC), then
BOOTCONTROL_OVERLAYS are not applied, but rcm_bootcontrol_overlay instead.

Which branch are you on?  The logic described does not seem to match https://github.com/OE4T/meta-tegra/blob/2bbc15ae8b5aa1dc570eae544cd15d83ce239a9a/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra-flash-helper.sh#L197-L202

rambrandt, 2025-02-25:
Not the latest, but close:
```
commit c7e77d1b8ef7377b8fa9064b1c800f0c354db927 (HEAD)
Author: Matt Madison <matt@madison.systems>
Date:   2025-01-11 09:05:56 -0800

    tegra-helper-scripts: fix RCM boot blob signing
```

I am not sure, but if L4TConfiguration-RootfsRedundancyLevelABEnable.dtbo is included or not, seems to make a difference here:

https://github.com/OE4T/meta-tegra/blob/2bbc15ae8b5aa1dc570eae544cd15d83ce239a9a/recipes-bsp/tegra-binaries/tegra-helper-scripts/tegra-flash-helper.sh#L717

rambrandt, 2025-02-25:
Thank you very much for looking into it.

As I mentioned in the issue description, it happens also with eMMC that the UEFI environment variable `RootfsRedundancyLevel` is 0, if I flash with `initrd-flash`.  It is `1` if I use `./doflash.sh`.

dwalkes, 2025-02-25:
> RootfsRedundancyLevel is 0, if I flash with `initrd-flash`

Understood, and if I do a clone of https://github.com/OE4T/tegra-demo-distro, then
* Setup build with
```
source setup-env --machine jetson-orin-nano-devkit-nvme
```
* Build image with `bitbake demo-image-base`
* tegraflash extract and `./initrd-flash`

On bootup I see 

```
root@jetson-orin-nano-devkit-nvme:~# nvbootctrl -t rootfs dump-slots-info
Current rootfs slot: A
Active rootfs slot: A
num_slots: 2
slot: 0,             retry_count: 3,             status: normal
slot: 1,             retry_count: 3,             status: normal
root@jetson-orin-nano-devkit-nvme:~#
```

And on reboot I see the same

So the problem is either something specific to your build setup or something specific to your MACHINE and if I were going to work on this I'd find out which as the next step.

rambrandt, 2025-02-26:
Thank you again very much for your help and your time.

I apologize for not testing with the latest version of the `tegra-flash-helper.sh` script from the beginning (just `initrd-flash`).

This commit seems to fix it:
```
commit f4fd64504fb58df3586f8484b8ddde9c8c3e8ea2
Author: Matt Madison <matt@madison.systems>
Date:   2025-01-19 06:10:55 -0800

    tegra-helper-scripts: restore main files after RCM boot signing

    The odmsign functions do the same post-processing for RCM boot signing
    as they do for main signing, so be sure to save and restore the main
    files after the second call for signing the RCM boot blob.

    Signed-off-by: Matt Madison <matt@madison.systems>
```

closed 2025-02-26 by rambrandt