OE4T/meta-tegra #1867: Build on ARM64 host

kraj, 2025-03-27:
**Describe the bug**

While building on x86 PC machine works ok, it fails to build on ARM64 machines.

Here is an error
```
| /home/khem/yoe/build/tmp/work/p3737_0000_p3701_0005-yoe-linux/tegra-minimal-initramfs/1.0/temp/run.do_image_cpio.3354548: 193: /home/khem/yoe/build/tmp/work/p3737_0000_p3701_0005-yoe-linux/tegra-minimal-initramfs/1.0/recipe-sysroot-native/usr/bin/tegra-flash/mkbootimg: Exec format error
| WARNING: exit code 126 from a shell command.
ERROR: Task (/home/khem/yoe/sources/meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_image_cpio) failed with exit code '1'
NOTE: Tasks Summary: Attempted 8129 tasks of which 2455 didn't need to be rerun and 1 failed.
````

To reproduce, one needs a ARM64 based PC or server and then do a usual build 

`bitbake tegra-minimal-initramfs`

madisongh, 2025-03-27:
Unfortunately, the tools NVIDIA provides us are binary-only and built for x86/x86-64. I've had success in the past with using binfmt_misc on ARM64 to have the programs run under qemu. Haven't tried that lately, though.

dwalkes, 2025-03-27:
I've seen tegraflash work on an RPI once upon a time. See https://github.com/orgs/OE4T/discussions/1574

kraj, 2025-03-28:
Fixed in container/Dockerfile for yoe as well.

closed 2025-03-28 by kraj