OE4T/meta-tegra #1875: dmesg boot errors

Lander0606, 2025-04-07:
Hi all,

When booting my Jetson Orin Nano with the image I created in Yocto (`main` branch for Yocto and meta-tegra), I encountered some (non-critical) errors in the boot logs regarding entries in the device tree:

```
/bus@0/cam_i2cmux/i2c@1/rbpcv2_imx219_c@10: Fixed dependency cycle(s) with /bus@0/host1x@13e00000/nvcsi@15a00000/channel@1
/bus@0/cam_i2cmux/i2c@0/rbpcv2_imx219_a@10: Fixed dependency cycle(s) with /bus@0/host1x@13e00000/nvcsi@15a00000/channel@0
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/processing-engine@2908000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/asrc@2910000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/amixer@290bb00
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/adx@2903b00
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/adx@2903a00
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/adx@2903900
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/adx@2903800
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/amx@2903300
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/amx@2903200
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/amx@2903100
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/amx@2903000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/mvc@290a200
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/mvc@290a000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/sfc@2902600
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/sfc@2902400
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/sfc@2902200
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/sfc@2902000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/dspk@2905100
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/dspk@2905000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/dmic@2904300
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/dmic@2904100
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/dmic@2904000
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/i2s@2901400
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/i2s@2901300
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/i2s@2901200
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/i2s@2901100
/bus@0/aconnect@2900000/ahub@2900800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800/admaif@290f000
/bus@0/aconnect@2900000/ahub@2900800/i2s@2901100: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/i2s@2901200: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/i2s@2901300: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/i2s@2901400: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/sfc@2902000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/sfc@2902200: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/sfc@2902400: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/sfc@2902600: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/amx@2903000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/amx@2903100: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/amx@2903200: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/amx@2903300: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/adx@2903800: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/adx@2903900: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/adx@2903a00: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/adx@2903b00: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/dmic@2904000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/dmic@2904100: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/dmic@2904300: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/dspk@2905000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/dspk@2905100: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/processing-engine@2908000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/mvc@290a000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/mvc@290a200: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/amixer@290bb00: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/admaif@290f000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
/bus@0/aconnect@2900000/ahub@2900800/asrc@2910000: Fixed dependency cycle(s) with /bus@0/aconnect@2900000/ahub@2900800
```

```
i2c i2c-0: of_i2c: modalias failure on /bus@0/i2c@3160000/prod-settings
i2c i2c-0: Failed to create I2C device for /bus@0/i2c@3160000/prod-settings
tegra-i2c 3180000.i2c: Adding to iommu group 1
i2c i2c-2: of_i2c: modalias failure on /bus@0/i2c@3180000/prod-settings
i2c i2c-2: Failed to create I2C device for /bus@0/i2c@3180000/prod-settings
tegra-i2c 31b0000.i2c: Adding to iommu group 1
[... 37 lines cut ...]
tegra186-emc 2c60000.external-memory-controller: sync_state() pending due to 15380000.nvjpg
tegra-mc 2c00000.memory-controller: sync_state() pending due to 15380000.nvjpg
tegra186-emc 2c60000.external-memory-controller: sync_state() pending due to 15540000.nvjpg
tegra-mc 2c00000.memory-controller: sync_state() pending due to 15540000.nvjpg
tegra186-emc 2c60000.external-memory-controller: sync_state() pending due to 15a50000.ofa
tegra-mc 2c00000.memory-controller: sync_state() pending due to 15a50000.ofa
tegra186-emc 2c60000.external-memory-controller: sync_state() pending due to 13e00000.host1x:vi0@15c00000
tegra-mc 2c00000.memory-controller: sync_state() pending due to 13e00000.host1x:vi0@15c00000
tegra186-emc 2c60000.external-memory-controller: sync_state() pending due to 13e00000.host1x:vi0-thi@15f00000
tegra-mc 2c00000.memory-controller: sync_state() pending due to 13e00000.host1x:vi0-thi@15f00000
tegra186-emc 2c60000.external-memory-controller: sync_state() pending due to 13e00000.host1x:vi1@14c00000
tegra-mc 2c00000.memory-controller: sync_state() pending due to 13e00000.host1x:vi1@14c00000
[... cut ...]
tegra-mc 2c00000.memory-controller: sync_state() pending due to 13e00000.host1x:vi1-thi@14f00000
```

For now, the system seems to work stable and fine but I'm not sure what about the I2C and SPI interface as these messages indicate some problems in the device tree.

The device tree of the system seems to be a combination of multiple snippets so I'm not sure where to look and what to change to solve these messages.

Branch: `master`
Machine: `Jetson Orin Nano Devkit 8GB`
Kernel:  `linux-yocto 6.12.19`

I'm looking forward for some context or possible solutions! Thanks in advance!

Lander0606, 2025-04-07:
Hi @ichergui,

Nice, thanks for your reply! Is there a neat way to change the device tree for the Jetson Orin Nano?

I found the `nvidia-kernel-oot-dtb` recipe, but it seems like this recipe only installs the compiled device tree. Is there another recipe which compiles the device tree and where it's possible to change the contents of it?

I want to experiment a bit on the above errors and change states of some devices.

Thanks in advance!

notabd7-deepshard, 2025-12-09:
bumping this issue ^
was this resolved? if it was can you provide link to fix please? Facing this exact issue which is causing spi to not work

closed 2026-09-27 by madisongh