OE4T/meta-tegra #1877: jetson orin AGX eMMC size wrong?

pauladam316, 2025-04-08:
Running into an issue trying to make a build for the jetson orin agx devkit where the rootfs on the eMMC only uses half of the 64gb available. Digging into the machine conf for the device (`agx-orin.inc`) I see

`EMMC_SIZE ?= "31276924928"`

which looks like this is configured for a 32gb eMMC. Is this a mistake to be corrected or is that the intentional behavior? I tried increasing it but that created build errors, as I understand it the flash partition xml needs to be updated as well, but its not clear exactly how that is done

ichergui, 2025-04-08:
That was not a mistake.
Please share the error logs so we can help.

pauladam316, 2025-04-09:
Here is the log I get when I try to modify the EMMC size
```
| [   5.2049 ]           55	worm                                	  129920	  130303
| [   5.2049 ]           56	BCT-boot-chain_backup               	  130304	  130431
| [   5.2049 ]           57	reserved_partition                  	  130432	  130559
| [   5.2049 ]           58	secondary_gpt_backup                	  130560	  130687
| [   5.2049 ]           59	B_VER                               	  130688	  130815
| [   5.2049 ]           60	A_VER                               	  130816	  130943
| [   5.2049 ] End sector for APP_b, expected at: 119537630, actual: 0
| [   5.2049 ]
| Error: Return value 4
| Command tegraparser_v2 --generategpt --pt flash.xml.bin
| cp: cannot stat 'signed/*': No such file or directory
| Welcome to Tegra Flash
| version 1.0.0
| Type ? or help for help and q or quit to exit
| Use ! to execute system commands
| 
| [   0.0348 ] tegrasign_v3.py --key None --getmode mode.txt
| [   0.0349 ] Assuming zero filled SBK key
| [   0.0341 ] tegra234-p3737-0000+p3701-0005-nv.dtb has no overlay_dtb files.
| [   0.0341 ] Concatenating dtbo files to tegra234-p3737-0000+p3701-0005-nv.dtb
| [   0.0347 ] Concatenating L4TConfiguration-rcmboot.dtbo,tegra234-p3737-0000+p3701-0000-dynamic.dtbo,tegra234-carveouts.dtbo,tegra-optee.dtbo,tegra234-p3737-camera-dual-imx274-overlay.dtbo,tegra234-p3737-camera-e3331-overlay.dtbo,tegra234-p3737-camera-e3333-overlay.dtbo,tegra234-p3737-camera-imx185-overlay.dtbo to tegra234-p3737-0000+p3701-0005-nv_overlay.dtb.updated
| [   0.0352 ] Concatenating bl dtb:(tegra234-p3737-0000+p3701-0005-nv_overlay.dtb.updated), to cpubl binary: rcmboot_uefi_jetson.bin
| [   0.0383 ] Get magic id
| [   0.0389 ] tegraparser_v2 --get_magic bootloader_stage2
| [   0.0396 ] CPBL
| [   0.0400 ] partition type bootloader_stage2, magic id = CPBL
| [   0.0440 ] tegrahost_v2 --chip 0x23 0 --align rcmboot_uefi_jetson_with_dtb_aligned.bin
| [   0.0456 ] tegrahost_v2 --chip 0x23 0 --magicid CPBL --appendsigheader rcmboot_uefi_jetson_with_dtb_aligned.bin zerosbk
| [   0.0461 ] adding BCH for rcmboot_uefi_jetson_with_dtb_aligned.bin
| [   0.1674 ] tegrasign_v3.py --key None --list rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin_list.xml --pubkeyhash pub_key.key --sha sha512
| [   0.1676 ] Assuming zero filled SBK key
| [   0.1780 ] Warning: pub_key.key is not found
| [   0.1778 ] tegrahost_v2 --chip 0x23 0 --updatesigheader rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin.encrypt rcmboot_uefi_jetson_with_dtb_aligned_sigheader.bin.hash zerosbk
| [   0.1907 ] Copying rcmboot_uefi_jetson_with_dtb_sigheader.bin.encrypt to /build/tmp/work/p3737_0000_p3701_0005-poky-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload
| [   0.1907 ] Signed file: /build/tmp/work/p3737_0000_p3701_0005-poky-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload/rcmboot_uefi_jetson_with_dtb_sigheader.bin.encrypt
| Signing with zerosbk ...
| rollback info is parsed and saved in rollback.bin
| WARNING: failed to find rollback config for " generic "
| pad ZEROs to the end of rollback.bin
| 
| signing images succeeded
| 
| The index file /build/tmp/work/p3737_0000_p3701_0005-poky-linux/tegra-uefi-capsules/36.3.0/tegra-uefi-capsules-36.3.0/bup-payload/signed/flash.idx is not found.
| WARNING: exit code 1 from a shell command.
ERROR: Task (/build/../work/meta-tegra/recipes-bsp/uefi/tegra-uefi-capsules_36.3.0.bb:do_compile) failed with exit code '1'
```

This was with trying to increase the `EMMC_SIZE` to 62553849856

Are you able to provide a bit more info on why its setup this way? if the jetson is configured with 64gb emmc, why is only half used?

Thanks!

ichergui, 2025-04-09:
Hi @pauladam316 

When redefining the `eMMC` size you need to update both `TEGRA_EXTERNAL_DEVICE_SECTORS` and `TEGRA_INTERNAL_DEVICE_SECTORS` otherwise the flashing process will failed.

You can use the following:   

```
EMMC_SIZE = "63350767616"  
ROOTFSPART_SIZE = "25769803776"  
TEGRA_EXTERNAL_DEVICE_SECTORS = "123731968"  
TEGRA_INTERNAL_DEVICE_SECTORS = "123731968" 
```

Let me know if that works

pauladam316, 2025-04-11:
Unfortunately I'm still getting the same error. 

Does it have to do with me having `USE_REDUNDANT_FLASH_LAYOUT_DEFAULT = "1"` Removing this fixes the build error but produces an image that doesn't boot. Its not clear to me exactly how it works, but I'm assuming if I'm changing the EMMC size I need to manually define the redundant partitions?

dwalkes, 2025-04-11:
> Does it have to do with me having USE_REDUNDANT_FLASH_LAYOUT_DEFAULT = "1" 

Possibly since this divides the default rootfs size in half, see https://github.com/OE4T/meta-tegra/wiki/Redundant-Rootfs-A-B-Partition-Support

What size is your rootfs?

The way to troubleshoot this is to look at the flash.xml files which are being parsed, and corresponding sizes/sector offsets to see which one is blowing out the end of the flash.  It looks like it's probably your APP/APP_b partition but I can't tell from the prints.  You might setup a minimally modified fork of tegra-demo-distro with the changes to reproduce, will be much easier to provide guidance in that case.

Regarding how to change the partition layout - you shouldn't need to do it AFAIK but there's an example at https://github.com/Trellis-Logic/kas-demos/blob/main/layers/meta-trellis-swupdate-rootfs-overlay-oe4t/recipes-bsp/tegra-binaries/tegra-storage-layout/external-flash.xml which shows the process.  Use other branches to see the process for the specific branch you are targeting.

pauladam316, 2025-04-14:
Update on this - I was able to resolve it!

@dwalkes you were right that it was the APP_b partition that was overrunning, checking the flash.xml confirmed this.

Looks like the values that @ichergui provided were good, but a different layer in my build was overwriting the `ROOTFSPART_SIZE`, fixing that resolved the problem and I was able to build and deploy with a 64gb emmc.

Thanks everyone for their help!

closed 2025-04-14 by pauladam316

ichergui, 2025-04-14:
@pauladam316 You're welcome 🙂