OE4T/meta-tegra #1892: ERROR: might be timeout in USB write. with TEGRA_SIGNING_ARGS set

arrow53, 2025-05-03:
**Describe the bug**
I'm seeing something similar to this
https://github.com/OE4T/meta-tegra/issues/1698

I see that it was fixed by @madisongh here https://github.com/OE4T/meta-tegra/pull/1800

but, when I try
```
./doflash.sh   -u /opt/oe/scarthgap/meta-crypt/rsa_priv.pem   -v /opt/oe/scarthgap/meta-crypt/sbk.key
```
I get
```
[   0.0351 ] Parsing partition layout
[   0.0354 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0370 ] Parsing partition layout
[   0.0373 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0380 ] mb1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --mb1_bin
[   0.0380 ] psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --psc_bl1_bin
[   0.0380 ] Boot Rom communication
[   0.0383 ] tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
[   0.0385 ] BR_CID: 0x80012344705DE8033400000003FF8100
[   0.0395 ] Sending bct_br
[   0.0395 ] Sending mb1
[   0.0399 ] ERROR: might be timeout in USB write.
Error: Return value 3

```

**To Reproduce**
I'm using `tegra-demo-distro` and the `scarthgap` branch. I am swapping out the latest `scarthgap` commit on `meta-tegra` just so I made sure to pick up the fix.

Any ideas to try would be greatly appreciated.

madisongh, 2025-05-04:
Based on that BR_CID, it looks like your device hasn't been fused for secure boot. You cannot use signed binaries to flash unsigned devices.

arrow53, 2025-05-04:
ok thanks @madisongh 
I went ahead and burned the fuses and the BR_CID changed but I still seem to be having issues
```
[   0.0190 ] Parsing partition layout
[   0.0194 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0214 ] Parsing partition layout
[   0.0217 ] tegraparser_v2 --pt secureflash.xml.tmp
[   0.0224 ] mb1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --mb1_bin
[   0.0224 ] psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed filename is from --psc_bl1_bin
[   0.0224 ] Boot Rom communication
[   0.0227 ] tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
[   0.0229 ] BR_CID: 0xE9012344705DE8033400000003FF8100
[   0.0239 ] Sending bct_br
[   0.0240 ] Sending mb1
[   0.0245 ] ERROR: might be timeout in USB write.
Error: Return value 3
Command tegrarcm_v2 --new_session --chip 0x23 0 --uid --download bct_br br_bct_BR.bct --download mb1 mb1_t234_prod_aligned_sigheader_encrypt.bin.signed --download psc_bl1 psc_bl1_t234_prod_aligned_sigheader_encrypt.bin.signed --download bct_mb1 mb1_bct_MB1_sigheader_encrypt.bct.signed
```
This is the doflash i'm using
```
./doflash.sh   -u rsa_priv.pem   -v sbk.key
```

arrow53, 2025-05-04:
I must be generating the keys or burning incorrectly as using the board with fuses blow and my generate keys in JetPack produces the same result.