OE4T/meta-tegra #1895: "Using device tree overlays" wiki

albertbu, 2025-05-07:
Hi, 
I am trying to add a separate dtb overlay, using the latest "scarthgap" branch. 
I add the KERNEL_DEVICETREE_APPLY_OVERLAYS as instructed by the wiki, but it seems that nothing happens. 
Digging in the meta-tegra, it seems that there is no provision for this feature there, but I see the hash in the current branch. 

```
aburbea@EXO-C-ALBERTB:~/projects-nvdia/XXXXXX/meta-tegra (scarthgap)$ git branch --contains 704382cedd1da1a98919936098da4742ee988a3b
  master
* scarthgap

```
I'm using yocto as linux provider. Any help please? 
Albert

dwalkes, 2025-05-07:
I think the best outline of this might be in https://github.com/orgs/OE4T/discussions/1455 and we probably need to fold that into https://github.com/OE4T/meta-tegra/wiki/Using-device-tree-overlays for r35 and later branches.

albertbu, 2025-05-28:
HI @Lander0606, 
mind to share the details?
Thanks

Lander0606, 2025-05-28:
Sure! If you use `devtool modify nvidia-kernel-oot`, the sources will be placed in `build/workspace/sources/nvidia-kernel-oot/`. You can then find the device tree files in `hardware/nvidia/t23x/nv-public`, which will be used to construct the final device tree for your device. Depending on what you want to change, you will need to make the change in a specific `dts` or `dtsi` file. I made some changes to `tegra234-p3768-0000+p3767.dtsi` to disable some pcie interfaces for my Jetson Orin Nano, and to `nv-soc/tegra234-soc-overlay.dtsi` to enable the Tegra Security Engine. You will need to use a command like `grep` to search through the device tree files to investigate what you need to change.

After you made some changes, go back to the `nvidia-kernel-oot` directory in `sources/` and commit your changes. You can then use `devtool finish nvidia-kernel-oot /path/to/your-layer` to copy your changes and patch to your own layer, or `git format-patch -1 HEAD` to create a patch file from your commit and copy it to your own layer.

Hope this helps! :)

dwalkes, 2026-01-21:
Hi @rytest-1214 thanks for the offer to help.  There are a lot of pages which could use some updates.

I did recently add https://github.com/OE4T/meta-tegra/wiki/Using-device-tree-overlays#example-out-of-tree-devicetree-in-tegra-demo-distro - you could add some additional examples there for other MACHINEs perhaps.

I've been working on a change to move wiki content into the source repo and into each repo branch, so one thing which could use some help is to split this content by Jetpack 6 vs jetpack 5 and make wiki pages dedicated to each.  This page is one of the pages which will be Jetpack/branch specific.  We could pull the Jetpack 5 specific content into the scarthgap-l4t-r35.x branch and leave the Jetpack 6 content in master/scarthgap.  Starting by preparing two wiki pages with different content would be great

You can see the work I've started for this at https://github.com/trellis-logic/meta-tegra and https://trellis-logic.github.io/meta-tegra/

That's just one idea, really anything you've learned in the process thus far would be useful to get documented in the wiki pages.  I'll move any wiki content into the source branches later this month.