OE4T/meta-tegra #1915: Mender update - Migration from yocto honister to scarthgap-l4t-r35

SylvainPastor, 2025-06-04:
I am currently working on an update to the yocto version with mender support.
Honister starting version towards scarthgap.

I need to keep the same layout (same sizes between APP, APP_b and UDA) so that my mender images can work on products already deployed for years and are compatible with SKUs `301` and `302`.

```
# NOTE:
# These values are pinned to maintain backward compatibility with SKU 301 devices
# originally deployed with Honister-based images.
# Changing them would break OTA updates and rollback for existing devices.
```

So I configure with the same sizes

local.conf

```
# Enable Mender logic for this build
INHERIT += "mender-full"
INHERIT:append:tegra = " tegra-mender-setup"
MYPRODUCT_MENDER_ENABLED = "1"

# Conditionally include Mender-specific providers
require conf/distro/include/myproduct-mender-preferred.inc
```

# Enable redundant root filesystem layout (A/B) to support atomic OTA updates with Mender.
# This switches the partition layout template to the *_rootfs_ab.xml version,
# ensuring the presence of both APP and APP_b partitions.
USE_REDUNDANT_FLASH_LAYOUT = "1"

myproduct-mender-preferred.inc:

```
# Adjust reserved space for APP/APP_b so UDA has room
TEGRA_MENDER_RESERVED_SPACE_MB = "1974"

# Allocate a larger persistent data partition
MENDER_DATA_PART_SIZE_MB_DEFAULT:tegra = "3072"
```

For my first test, I’m trying to apply my layout on a j`etson-xavier-nx-devkit-emmc` before trying with my machines.

I am encountering the following error:

```
[0024.302] W> RATCHET: MB1 binary ratchet value 4 is larger than ratchet level 2 from HW fuses.
[0024.310] I> MB1 (prd-version: 2.6.0.0-t194-41334769-cab45716)
[0024.316] I> Boot-mode: Platform RCM
[0024.319] I> Platform: Silicon
[0024.322] I> Chip revision : A02P
[0024.325] I> Bootrom patch version : 15 (correctly patched)
[0024.330] I> ATE fuse revision : 0x200
[0024.333] I> Ram repair fuse : 0x0
[0024.337] I> Ram Code : 0x1
[0024.339] I> rst_source: 0x0, rst_level: 0x0
[0024.344] I> USB configuration success
[0026.920] I> mb2 image downloaded
[0027.091] I> Recovery boot mode 0
[0027.097] I> UPHY full init done
[0027.104] I> MB1 done

[0027.109] I> Welcome to MB2(TBoot-BPMP) Applet (version: default.t194-mobile-dba32e6d)
[0027.116] I> DMA Heap @ [0x40020000 - 0x40065800]
[0027.121] I> Default Heap @ [0xd486400 - 0xd48a400]
[0027.125] W> Profiler not initialized
[0027.237] I> sdmmc DDR50 mode
[0027.244] I> QSPI-0l initialized successfully
[0027.251] E> CLK_RST: instance 6 not found in module 44.
[0027.256] E> MPhy CAR configuration failed error = 1747992077
[... 262 lines cut ...]
[0274.569] I> Writing kernel-dtb_b partition.
[0274.603] I> Writing esp partition.
[0279.005] I> Writing UDA partition.
[0279.037] E> NV3P_SERVER: Could not write 1048576 bytes
```.

I feel like I lost a bit of space with Scarthgap:

Adds several mandatory EFI partitions:
```
esp, esp_alt,
recovery, recovery_alt, RECROOTFS
uefi_variables, uefi_ftw
```

Approximately 300 to 400 MiB more taken up in the layout compared to honister.

I tried truncating the size of my image but nothing works.

always the same error:

```
[0279.005] I> Writing UDA partition.
[0279.037] E> NV3P_SERVER: Could not write 1048576 bytes.
```

SylvainPastor, 2025-06-04:
Here is the distribution of my partitions (Read from flash.xml.in)
```
eMMC capacity: 15758000128 bytes (14.68 GiB)

master_boot_record                      512 bytes
primary_gpt                           19968 bytes
APP                              5233442816 bytes
APP_b                            5233442816 bytes
kernel                             67108864 bytes
kernel-dtb                           458752 bytes
reserved_for_chain_A_user          34123776 bytes
secure-os_b                         2621440 bytes
eks_b                                 65536 bytes
adsp-fw_b                           1048576 bytes
rce-fw_b                            1048576 bytes
sce-fw_b                            1048576 bytes
bpmp-fw_b                           1572864 bytes
bpmp-fw-dtb_b                       1048576 bytes
kernel_b                           67108864 bytes
kernel-dtb_b                         458752 bytes
reserved_for_chain_B_user          34078720 bytes
recovery                           83886080 bytes
recovery-dtb                         524288 bytes
RECROOTFS                         104857600 bytes
esp                                67108864 bytes
recovery_alt                       83886080 bytes
recovery-dtb_alt                     524288 bytes
esp_alt                            67108864 bytes
UDA                                   18432 bytes
secondary_gpt                       IGNORED (0xFFFFFFFFFFFFFFFF)

--- Summary ---
Total used space      : 11086612480 bytes
Remaining space       : 4671387648 bytes
Remaining space (GiB) : 4.35 GiB
Utilization           : 70.36%
✅ OK: Partitions fit within eMMC capacity.
```

SylvainPastor, 2025-06-04:
The distribution with my `honister` version which worked

```
eMMC capacity: 15758000128 bytes (14.68 GiB)

master_boot_record                      512 bytes
primary_gpt                           19968 bytes
APP                              5233442816 bytes
kernel                             67108864 bytes
kernel_b                           67108864 bytes
kernel-dtb                           458752 bytes
kernel-dtb_b                         458752 bytes
recovery                           66060288 bytes
recovery-dtb                         524288 bytes
kernel-bootctrl                      262144 bytes
kernel-bootctrl_b                    262144 bytes
RECROOTFS                         314572800 bytes
APP_b                            5233442816 bytes
UDA                                   18432 bytes
secondary_gpt                       IGNORED (0xFFFFFFFFFFFFFFFF)

--- Summary ---
Total used space      : 10983741440 bytes
Remaining space       : 4774258688 bytes
Remaining space (GiB) : 4.45 GiB
Utilization           : 69.70%
✅ OK: Partitions fit within eMMC capacity.
```

SylvainPastor, 2025-06-04:
I did a series of tests and here are the results:

UDA partition released by Yocto:
```
<partition name="UDA" type="data">
      <allocation_policy> sequential </allocation_policy>
      <filesystem_type> basic </filesystem_type>
      <size> 18432 </size>
      <file_system_attribute> 0 </file_system_attribute>
      <allocation_attribute> 0x808 </allocation_attribute>
      <percent_reserved> 0 </percent_reserved>
     <filename> DATAFILE </filename>
      <description> **Required.** Automatically takes all remaining space on the device except that
        occupied by the `secondary_gpt` partition. Allocation attribute must be set to 0x808.
        May be mounted and used to store user data. 
     </description>
  </partition>
```
Test 1: `Fixed UDA (without resize-to-fit)` / 4 GiB
```
<partition name="UDA" type="data">
      <allocation_policy> sequential </allocation_policy>
      <filesystem_type> basic </filesystem_type>
      <size> 4294967296 </size>
      <file_system_attribute> 0 </file_system_attribute>
      <allocation_attribute> 0x8 </allocation_attribute>
      <percent_reserved> 0 </percent_reserved>
     <filename> DATAFILE </filename>
      <description> **Required.** Automatically takes all remaining space on the device except that
        occupied by the `secondary_gpt` partition. Allocation attribute must be set to 0x808.
        May be mounted and used to store user data. 
     </description>
  </partition>
```
Result:
```
[  20.3472 ] End sector for UDA, expected at: 30777310, actual: 0
[  20.3472 ] 
Error: Return value 4
Command tegraparser_v2 --storageinfo storage_info.bin --generategpt --pt flash.xml.bin
```

Test 2: Dynamic UDA (resize-to-fit, without size) / size attribut deleted
```
<partition name="UDA" type="data">
      <allocation_policy> sequential </allocation_policy>
      <filesystem_type> basic </filesystem_type>
      <file_system_attribute> 0 </file_system_attribute>
      <allocation_attribute> 0x800 </allocation_attribute>
      <percent_reserved> 0 </percent_reserved>
     <filename> DATAFILE </filename>
      <description> **Required.** Automatically takes all remaining space on the device except that
        occupied by the `secondary_gpt` partition. Allocation attribute must be set to 0x808.
        May be mounted and used to store user data. 
     </description>
  </partition>
```
Result:
```
[0327.333] E> Cannot find partition UDA
[0327.334] E> NV3P_SERVER: Failed to open partition UDA.
```

Test 3: neither logic nor resize
```
<partition name="UDA" type="data">
      <allocation_policy> sequential </allocation_policy>
      <filesystem_type> basic </filesystem_type>
      <size> 4294967296 </size>
      <file_system_attribute> 0 </file_system_attribute>
      <allocation_attribute> 0x0 </allocation_attribute>
      <percent_reserved> 0 </percent_reserved>
     <filename> DATAFILE </filename>
      <description> **Required.** Automatically takes all remaining space on the device except that
        occupied by the `secondary_gpt` partition. Allocation attribute must be set to 0x808.
        May be mounted and used to store user data. 
     </description>
  </partition>
```
Result:
```
Error: Return value 4
Command tegraparser_v2 --storageinfo storage_info.bin --generategpt --pt flash.xml.bin
```

Test 4: NVIDIA method: Remove fliename attribut
```
<partition name="UDA" type="data">
      <allocation_policy> sequential </allocation_policy>
      <filesystem_type> basic </filesystem_type>
      <size> 4294967296 </size>
      <file_system_attribute> 0 </file_system_attribute>
      <allocation_attribute> 0x808 </allocation_attribute>
      <percent_reserved> 0 </percent_reserved>
      <description> **Required.** Automatically takes all remaining space on the device except that
        occupied by the `secondary_gpt` partition. Allocation attribute must be set to 0x808.
        May be mounted and used to store user data. 
     </description>
  </partition>
```

Result: 

```
[0322.220] I> Writing kernel-dtb_b partition.
[0322.261] I> Writing esp partition.
[0327.342] I> Writing BCT partition.
[0327.559] I> Writing BCT-boot-chain_backup partition.
[0327.902] I> Writing MB1_BCT partition.
[0328.226] I> Writing MB1_BCT_b partition.
[0328.559] I> Writing MEM_BCT partition.
[0330.065] I> Writing MEM_BCT_b partition.
[0331.762] I> Rebooting : reset-coldboot
```
I am continuing my investigations to have a fully operational mender

madisongh, 2025-06-07:
I'm not exactly sure what you're asking here. As you can see, NVIDIA added several partitions to the eMMC with L4T R35.x - the UEFI ESP, and most of the `_b` partitions for boot firmware that no longer fit in the small SPI flash on the Xavier NX - so you may end up having to resize your own partitions to account for there being less available space. Some of the partitions in the stock NVIDIA layout also have fixed start locations set in the XML file, which you may need to adjust if you change the sizes of partitions that precede them. The `0x808` allocation attribute marks the partition as "fill to end," so the UDA partition should consume as much remaining space as it can. Unfortunately, with R35.x, that's only about 200,000 sectors (100MiB) with the stock NVIDIA layout, as you found.

closed 2025-11-30 by ichergui