OE4T/meta-tegra #1944: Error getting storage info when running `./doflash` command - "Applet/MB2 is not running on device"

Onizuka09, 2025-07-22:
I have an nvidia jeston orin nano 4GB 
When running the `./doflash` command after building either `demo-image-base` or a custom image, I encounter the following error:
```text
[ 13.6824 ] tegrarcm_v2 --chip 0x23 0 --oem platformdetails storage storage_info.bin
[ 13.6843 ] Applet/MB2 is not running on device.
[ 2047.4002 ] Error: failed to get storage info
```
how can i fix this ?

madisongh, 2025-07-25:
Since the Orin Nano modules have no on-board eMMC for storage, you have to use the [initrd flashing method](https://github.com/OE4T/meta-tegra/wiki/initrd-flashing-support).

Onizuka09, 2025-07-27:
i have done that, so here is what, i  followed the `tegra-demo` instructions, i have followed the `initrd-method` to to set up my host environment but i get this `kernel-panic` error on the serial monitor 
there seem to be an issue with the init script 

```
....
   3.775388] fusb301 1-0025: fusb301_work_handler: int_sts[0x05]
[    3.775952] fusb301 1-0025: sts[0x2b], type[0x08]
[    3.775961] fusb301 1-0025: fusb_update_state: 6
[    3.876276] spi-tegra114 3210000.spi: Adding to iommu group 1
[    3.879286] spi-tegra114 3230000.spi: Adding to iommu group 1
[    3.929904] tegra234-aon c000000.aon: Adding to iommu group 2
[    3.932334]  c000000.aon:hsp: probed
[    3.932613] tegra234-aon c000000.aon: init done
[    3.976224] tegra-soc-hwpm f100000.hwpm: Adding to iommu group 3
[    4.082391] tegra-qspi 3270000.spi: Adding to iommu group 4
[    4.083814] tegra-qspi 3270000.spi: Prod config not found for QSPI: -19
[    4.086530] spi-nor spi2.0: mx25u51279g (65536 Kbytes)
[    4.187430] nvvrs_pseq 4-003c: NVVRS Vendor ID: 0x9
[    4.187775] nvvrs_pseq 4-003c: NVVRS Model Rev: 0x82
[    4.188455] nvvrs_pseq 4-003c: NVVRS PSEQ probe successful
[    4.426635] Kernel panic - not syncing:
[    4.426650] Attempted to kill init! exitcode=0x00000100
[    4.426661] CPU: 1 PID: 1 Comm: init Tainted: G           O      5.15.148-1
[    4.426667] Hardware name: NVIDIA NVIDIA Jetson Orin Nano Engineering Refe4
[    4.426675] Call trace:
[    4.426677]  dump_backtrace+0x0/0x1d0
[    4.426725]  show_stack+0x34/0x4c
[    4.426731]  dump_stack_lvl+0x64/0x80
[    4.426759]  dump_stack+0x18/0x34
[    4.426761]  panic+0xd4/0x3a0
[    4.426764]  do_exit+0x9c4/0xa00
[    4.426782]  do_group_exit+0x44/0xa4
[    4.426784]  __wake_up_parent+0x0/0x50
[    4.426787]  invoke_syscall+0x5c/0x130
[    4.426797]  el0_svc_common.constprop.0+0x64/0x110
[    4.426801]  do_el0_svc+0x3c/0xa0
[    4.426804]  el0_svc+0x20/0x60
[    4.426819]  el0t_64_sync_handler+0x108/0x114
[    4.426820]  el0t_64_sync+0x1a4/0x1a8
```

Onizuka09, 2025-07-29:
i got it fixed i was using the wrong machine type, the correct one is `MACHINE ??= "p3768-0000-p3767-0004"`