OE4T/meta-tegra #1952: Persistent Login Issue: Custom User Creation Fails on NVIDIA Jetson AGX Orin with Yocto

felixkarevo, 2025-07-30:
## **Summary**
Despite extensive debugging and multiple attempted fixes, a custom user cannot log in on a Yocto-built image for NVIDIA Jetson AGX Orin. The system only shows a default `jetson` user at boot, and login attempts with the custom user fail via both GUI and TTY.

## **Environment**
- **Hardware**: NVIDIA Jetson AGX Orin 64GB (p3737-0000-p3701-0005)
- **Yocto Version**: Walnascar (5.2)
- **Meta-tegra Layer**: Latest compatible version
- **Build System**: Custom meta layer
- **Distribution**: Custom distro based on Poky
- **Init System**: systemd

## **Expected Behavior**
- Custom user should be created with specific UID
- User should be able to log in with configured password
- User should have sudo privileges and be member of required groups
- No `jetson` user should exist (this is a custom image, not NVIDIA's default)

## **Actual Behavior**
- Only `jetson` user appears at login screen
- Custom user login fails on both GUI and TTY2
- Custom user configuration appears to be ignored

## **Troubleshooting Performed**

### **1. Image Analysis via Loop Mount**
```bash
sudo mount -o loop custom-jetson-image.rootfs.ext4 /mnt/rootfs
```

**✅ User Creation Confirmed:**
```bash
$ cat /mnt/rootfs/etc/passwd | grep customuser
customuser:x:1200:1200::/home/customuser:/bin/bash
```

**✅ Password Hash Confirmed (SHA512):**
```bash
$ sudo cat /mnt/rootfs/etc/shadow | grep customuser
customuser:$6$salt$<SHA512_HASH>:15069:0:99999:7:::
```

**❌ Missing Group Memberships:**
```bash
$ cat /mnt/rootfs/etc/group | grep -E "(customuser|sudo|docker)"
sudo:x:27:                    # customuser NOT in sudo group
docker:x:993:customuser       # customuser in docker group ✓
customuser:x:1200:            # customuser primary group ✓
```

**❌ Missing Sudoers File:**
```bash
$ sudo ls -la /mnt/rootfs/etc/sudoers.d/
total 8
drwxr-x---  2 root root 4096 Apr  6  2011 .
drwxr-xr-x 44 root root 4096 Apr  6  2011 ..
# Empty - custom sudoers file missing!
```

**❌ Missing PAM Service Files:**
```bash
$ ls -la /mnt/rootfs/etc/pam.d/ | grep -E "(login|sshd)"
# No output - login and sshd PAM files missing!
```

### **2. PAM Configuration Analysis**

**✅ PAM Hash Support Confirmed:**
```bash
$ cat /mnt/rootfs/etc/pam.d/common-password
password [success=1 default=ignore] pam_unix.so sha512
```

**✅ PAM Authentication Chain:**
```bash
$ cat /mnt/rootfs/etc/pam.d/common-auth
auth [success=1 default=ignore] pam_unix.so nullok_secure
```

## **Root Cause Analysis**

### **🔍 Critical Issue: `pkg_postinst_ontarget` Not Executing**

The custom user recipe uses `pkg_postinst_ontarget()` to:
1. Install sudoers file to `/etc/sudoers.d/customuser`
2. Add user to system groups (`sudo`, `video`, `audio`, `adm`, `render`)
3. Install PAM service files (`/etc/pam.d/login`, `/etc/pam.d/sshd`)
4. Set root password

**Evidence that postinst is NOT running:**
- Sudoers directory is empty
- User not in `sudo` group
- PAM service files missing
- Group memberships incomplete

### **🔍 Suspected Cause: Missing `run-postinsts` Package**

Yocto's `pkg_postinst_ontarget` scripts require the `run-postinsts` systemd service to execute on first boot. This service may not be included in the custom image.

## **Current Recipe Configuration**

### **meta-custom/recipes-support/custom-user/custom-user_1.0.bb**
```bitbake
SUMMARY = "Custom user account management"
DESCRIPTION = "Creates and configures the custom user account with sudo and docker privileges."
LICENSE = "MIT"
LIC_FILES_CHKSUM = "file://${COMMON_LICENSE_DIR}/MIT;md5=0835ade698e0bcf8506ecda2f7b4f302"

SRC_URI = "file://custom-sudoers \
           file://login \
           file://sshd"

S = "${WORKDIR}/sources"
UNPACKDIR = "${S}"

# Inherit useradd class for proper user creation
inherit useradd

# Ensure sudo is available for the custom user
RDEPENDS:${PN} = "sudo"

# Set up user creation parameters
USERADD_PACKAGES = "${PN}"

# Create custom user with password hash (compatible with PAM SHA512 configuration)
# Note: Only add to groups that we create ourselves, system groups will be added in postinst
USERADD_PARAM:${PN} = "-u 1200 -d /home/customuser -m -s /bin/bash -G docker,weston-launch -p '\$6\$salt\$<SHA512_HASH>' customuser"

# Create required groups
GROUPADD_PARAM:${PN} = "-g 999 docker; -r weston-launch"

# System groups (sudo, video, audio, adm, render) will be added in postinst when they exist

do_install() {
    # Install the sudoers file content to a temp location
    install -d ${D}${datadir}/${PN}
    install -m 0440 ${S}/custom-sudoers ${D}${datadir}/${PN}/custom-sudoers
    
    # Install PAM service files
    install -m 0644 ${S}/login ${D}${datadir}/${PN}/login
    install -m 0644 ${S}/sshd ${D}${datadir}/${PN}/sshd
}

FILES:${PN} = "${datadir}/${PN}/custom-sudoers ${datadir}/${PN}/login ${datadir}/${PN}/sshd"

# Install the sudoers file on target to avoid directory ownership conflicts
pkg_postinst_ontarget:${PN}() {
    #!/bin/sh
    # Copy the sudoers file to the final location on target device
    cp ${datadir}/${PN}/custom-sudoers /etc/sudoers.d/customuser
    chmod 0440 /etc/sudoers.d/customuser
    chown root:root /etc/sudoers.d/customuser
    
    # Install PAM service files for login and SSH
    cp ${datadir}/${PN}/login /etc/pam.d/login
    cp ${datadir}/${PN}/sshd /etc/pam.d/sshd
    chmod 0644 /etc/pam.d/login /etc/pam.d/sshd
    chown root:root /etc/pam.d/login /etc/pam.d/sshd
    
    # Set root password (using same hash format as custom user)
    echo 'root:$6$salt$<SHA512_HASH>' | chpasswd -e
    
    # Add user to system groups that should exist on target
    usermod -a -G sudo customuser || true
    usermod -a -G video customuser || true
    usermod -a -G audio customuser || true
    usermod -a -G adm customuser || true
    usermod -a -G render customuser || true
    
    # Add user to display manager groups if they exist
    if [ -e /etc/gdm3/custom.conf ]; then
        usermod -a -G gdm customuser || true
        usermod -a -G video gdm || true
    fi
    
    if [ -e /lib/systemd/system/lightdm.service ]; then
        usermod -a -G lightdm customuser || true
        usermod -a -G video lightdm || true
    fi
    
    # Create NVIDIA autoconfig file (following NVIDIA's approach)
    mkdir -p /etc/nv
    touch /etc/nv/nvautoconfig
}

pkg_prerm:${PN}() {
    #!/bin/sh
    # Remove the sudoers file when package is removed
    rm -f /etc/sudoers.d/customuser
}
```

### **Image Recipe**
```bitbake
# Custom utilities
IMAGE_INSTALL:append = " \
    custom-system-info \
    custom-user \
    custom-security \
"
```

## **Attempted Fixes**

### **✅ Fix 1: Hash Format Compatibility**
- **Issue**: User had MD5 hash but PAM expected SHA512
- **Solution**: Updated password hash to SHA512 format
- **Result**: Hash now matches PAM configuration

### **✅ Fix 2: Added PAM Service Files**
- **Issue**: Missing `/etc/pam.d/login` and `/etc/pam.d/sshd`
- **Solution**: Created proper PAM service configurations
- **Result**: Files included in recipe, should be installed by postinst

### **✅ Fix 3: Group Management Strategy**
- **Issue**: Build-time group conflicts
- **Solution**: Create only custom groups at build time, add system groups in postinst
- **Result**: Build succeeds without group errors

### **❌ Fix 4: Postinst Execution (FAILED)**
- **Issue**: `pkg_postinst_ontarget` not executing on first boot
- **Suspected Cause**: Missing `run-postinsts` package
- **Status**: UNRESOLVED

## **Questions**

1. **Is `run-postinsts` automatically included in systemd-based images?**
2. **How can I verify that `pkg_postinst_ontarget` scripts are scheduled to run?**
3. **Could the NVIDIA meta-tegra layer be interfering with user creation?**
4. **Why does a `jetson` user appear instead of the custom user?**
5. **Are there Jetson-specific requirements for user creation that I'm missing?**

## **References**
- [Yocto Project Manual - Post-Installation Scripts](https://docs.yoctoproject.org/dev-manual/new-recipe.html#post-installation-scripts)
- [NVIDIA LDAP Authentication Guide](https://docs.nvidia.com/networking/display/nvidianvosusermanualforinfinibandswitchesv25024002/ldap+authentication+and+authorization)

felixkarevo, 2025-07-31:
Hi @kekiefer , thank you for your advice. I extraxt (tar -xf) my custom-tar.zst built by bitbake/yocto. Then I run sudo ./doflash.sh. Which in turn runs ./tegra-flash-helper.sh. 

I had a look at the files (`tegra-flash-helper.sh`, `initrd-flash`, etc.), **none of them create users or modify the rootfs**. The flash process only handles:
- Hardware initialization
- Partition layout
- Bootloader configuration
- Binary flashing

The "jetson" user is not being created by the flashing process.

Based on some more research I did in the meantime I tried to implement some additional fixes.

#### **1. Simplified PAM Configuration**
**Problem**: Original PAM files used `@include common-auth` which may not exist on target.

**Solution**: Created minimal, self-contained PAM files:
```
# /etc/pam.d/login and /etc/pam.d/sshd
auth       required   pam_unix.so shadow nullok
account    required   pam_unix.so
password   required   pam_unix.so shadow nullok
session    required   pam_unix.so
```

#### **2. Package Configuration Improvements**
```bitbake
# Ensure the package is marked for installation
ALLOW_EMPTY:${PN} = "0"
```

I also verified the image build again. It seems that all components are correct. 

```bash
# Custom user exists in filesystem
$ grep customuser /mnt/etc/passwd
customuser:x:1200:1200::/home/customuser:/bin/bash

# Package files correctly installed
$ ls -la /mnt/usr/share/custom-user/
-r--r-----  1 root root  137 Apr  6  2011 custom-sudoers
-rw-r--r--  1 root root  202 Apr  6  2011 login
-rw-r--r--  1 root root  201 Apr  6  2011 sshd

# run-postinsts service enabled
$ ls -la /mnt/etc/systemd/system/sysinit.target.wants/run-postinsts.service
lrwxrwxrwx 1 root root 45 Apr  6  2011 -> /usr/lib/systemd/system/run-postinsts.service

#STILL AN ISSUE

# Sudoers directory still empty (expected - postinst runs on target)
$ ls -la /mnt/etc/sudoers.d/
total 8
drwxr-x---  2 root root 4096 Apr  6  2011 .
drwxr-xr-x 44 root root 4096 Apr  6  2011 ..
```

The current status after flashing the image is:
- Still only see "jetson" user at login screen
- Cannot login with custom user (GUI and tty2)
- Custom user not visible at login screen

1. **I still do not understand where the "jetson" user comes from???** 
   - It's NOT in `/etc/passwd` in the built filesystem
   - It's NOT created by flash scripts
   - **Is it hardcoded in UEFI/bootloader/display manager?**

2. **Is there a Jetson-specific login manager override?**
   - Could NVIDIA's UEFI be showing a default "jetson" user regardless of filesystem contents?
   - Are there Jetson-specific display manager configurations?

3. **Are `pkg_postinst_ontarget` scripts executing on Jetson?**
   - `run-postinsts.service` is enabled but may not be running
   - Could there be Jetson-specific boot sequence issues?

felixkarevo, 2025-08-01:
@dwalkes thank you for the feedback. I will first try this repo: [https://github.com/YoeDistro/yoe-distro](https://github.com/YoeDistro/yoe-distro). If this does not work, I will use the tegra-demo-distro as you suggested. In any case, I will post it here when it was successful.

felixkarevo, 2025-08-06:
Surprise... You pointed me into the right direction. The issue was not connected to Yocto (`usermod`, etc.) but to the way I flashed the image. I was always able to build, flash and boot. However, after the boot I always saw a ubuntu-gnome-like GUI with a user "jetson" that I had not created. 

The problem was the M.2/nvme harddisk on my jetson. This caused issues during the flashing and booting. I did not manage to flash the M.2/nvme harddisk. I tried different approaches but always failed. If you have some advice on how to flash a M.2/nvme drive, please let me know. Is it possible or recommended to have the operating system on a M.2/nvme harddisk? Or should it only be used as device to store data?  

Long story short: I removed the M.2/nvme device from the AGX Orin and flashed the yocto files onto the emmc chip. This approach worked. I was able to login with the user "root" without any password and I was also able to login with my custom user.

dwalkes, 2025-08-06:
>  If you have some advice on how to flash a M.2/nvme drive, please let me know.

See https://github.com/OE4T/meta-tegra/wiki/initrd-flashing-support for instructions to build an image for NVMe and use initrd-flash to flash it.

closed 2025-08-06 by dwalkes