OE4T/meta-tegra #1972: libnvparsers.so incorrectly linked

HuysArthur, 2025-08-26:
I'm experiencing issues were my c++ code can't find `lnvparsers` in my SDK created by Yocto. When debugging I saw there was a file `/opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so` but the link here doesn't seem correct. Here is some proof of how the dir looks like when unpacked from the script:
```
root@2e3e1f688a27:/workspaces/toadi-main# find /opt/eeve/1.0.0/sysroots/ -name "libnvparsers.so*"
/opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so.10
/opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so
root@2e3e1f688a27:/workspaces/toadi-main# ldd /opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so
ldd: /opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so: No such file or directory
root@2e3e1f688a27:/workspaces/toadi-main# ldd /opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so.10
ldd: /opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so.10: No such file or directory
root@2e3e1f688a27:/workspaces/toadi-main# file /opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so
/opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so: broken symbolic link to libnvparsers.so.10.3.0
root@2e3e1f688a27:/workspaces/toadi-main# file /opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so.10
/opt/eeve/1.0.0/sysroots/armv8a-poky-linux/usr/lib/libnvparsers.so.10: broken symbolic link to libnvparsers.so.10.3.0
```

The thing I added to my yocto config is this: `TOOLCHAIN_TARGET_TASK:append = " tegra-mmapi-dev tensorrt-core-dev"`

Is this a bug i'm experiencing or do I need to setup something else?

HuysArthur, 2025-08-26:
Apparently libnvparsers.so has been removed for TRT-10.x, so I think these lines should be removed from the bitbake:

```
ln -s libnvparsers.so.${BASEVER} ${D}${libdir}/libnvparsers.so.${MAJVER}
ln -s libnvparsers.so.${BASEVER} ${D}${libdir}/libnvparsers.so
```

Not that that was causing the issue, but it made it a bit confusing. I just had to remove `-lnvparsers` from a .cmake

ichergui, 2025-08-26:
The build instructions matter to me. So, share them
Specifically if you made changes to the `conf/local.conf` file

madisongh, 2025-09-29:
Thanks for the report, @HuysArthur !  I've queued up a PR to remove the broken symlinks.